facebookresearch/home-robot #56: Steps to install home_robot core and running hector slam

vidhiJain, 2023-02-17:
## 🐛 Bug

I tried setting it up and am running into errors in home_robot install and running hector slam.

## Steps to Reproduce

Steps to reproduce the behavior:
- Errors in testing home_robot install: What version of `sophus` is required? 
```
(home_robot) hello-robot@stretch-re2-2003:~/home-robot/src/home_robot$ python -m home_robot.nodes.goto_controller
/home/hello-robot/mambaforge/envs/home_robot/bin/python: Error while finding module specification for 'home_robot.nodes.goto_controller' (ModuleNotFoundError: No module named 'home_robot.nodes')
(home_robot) hello-robot@stretch-re2-2003:~/home-robot/src/home_robot$
(home_robot) hello-robot@stretch-re2-2003:~/home-robot/src/home_robot$ python -m home_robot.client.local_hello_robot
Traceback (most recent call last):
  File "/home/hello-robot/mambaforge/envs/home_robot/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/hello-robot/mambaforge/envs/home_robot/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/hello-robot/home-robot/src/home_robot/home_robot/client/local_hello_robot.py", line 16, in <module>
    import sophus as sp
ModuleNotFoundError: No module named 'sophus'
```
- Error in running hector slam: How to specify the launch files for hector slam?
```
(base) hello-robot@stretch-re2-2003:~/catkin_ws$ roslaunch home_robot startup_stretch_hector_slam.launch
RLException: [startup_stretch_hector_slam.launch] is neither a launch file in package [home_robot] nor is [home_robot] a launch file name
The traceback for the exception was written to the log file.
```

## Expected behavior
Instructions for checking correct setup of `home_robot` core package and running hector-slam.

## Additional context
NA

vidhiJain, 2023-03-03:
Error after updates:
```
hello-robot@stretch-re2-2003:~/catkin_ws/src/home_robot_hw/launch$ roslaunch home_robot_hw startup_stretch_hector_slam.launch
... logging to /home/hello-robot/.ros/log/85b825f6-b9f2-11ed-80a5-b31cbd894d96/roslaunch-stretch-re2-2003-12589.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.26.172.112:43585/

SUMMARY
========

PARAMETERS
 * /aggregator_node/analyzers/hardware/analyzers/Arm/find_and_remove_prefix: ['stretch_driver:...
 * /aggregator_node/analyzers/hardware/analyzers/Arm/path: Arm
 * /aggregator_node/analyzers/hardware/analyzers/Arm/type: diagnostic_aggreg...
 * /aggregator_node/analyzers/hardware/analyzers/Base/find_and_remove_prefix: ['stretch_driver:...
 * /aggregator_node/analyzers/hardware/analyzers/Base/path: Base
 * /aggregator_node/analyzers/hardware/analyzers/Base/type: diagnostic_aggreg...
 * /aggregator_node/analyzers/hardware/analyzers/EndOfArm/find_and_remove_prefix: ['stretch_driver:...
 * /aggregator_node/analyzers/hardware/analyzers/EndOfArm/path: EndOfArm
 * /aggregator_node/analyzers/hardware/analyzers/EndOfArm/type: diagnostic_aggreg...
 * /aggregator_node/analyzers/hardware/analyzers/Head/find_and_remove_prefix: ['stretch_driver:...
 * /aggregator_node/analyzers/hardware/analyzers/Head/path: Head
 * /aggregator_node/analyzers/hardware/analyzers/Head/type: diagnostic_aggreg...
 * /aggregator_node/analyzers/hardware/analyzers/Lift/find_and_remove_prefix: ['stretch_driver:...
 * /aggregator_node/analyzers/hardware/analyzers/Lift/path: Lift
 * /aggregator_node/analyzers/hardware/analyzers/Lift/type: diagnostic_aggreg...
 * /aggregator_node/analyzers/hardware/analyzers/Pimu/find_and_remove_prefix: ['stretch_driver:...
 * /aggregator_node/analyzers/hardware/analyzers/Pimu/path: Pimu
 * /aggregator_node/analyzers/hardware/analyzers/Pimu/type: diagnostic_aggreg...
 * /aggregator_node/analyzers/hardware/analyzers/Wacc/find_and_remove_prefix: ['stretch_driver:...
 * /aggregator_node/analyzers/hardware/analyzers/Wacc/path: Wacc
 * /aggregator_node/analyzers/hardware/analyzers/Wacc/type: diagnostic_aggreg...
 * /aggregator_node/analyzers/hardware/path: Hardware
 * /aggregator_node/analyzers/hardware/type: diagnostic_aggreg...
 * /aggregator_node/path: Stretch
 * /aggregator_node/type: diagnostic_aggreg...
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
[... 212 lines cut ...]
    from control_msgs.msg import FollowJointTrajectoryAction
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 914, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1407, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1379, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1525, in find_spec
  File "<frozen importlib._bootstrap_external>", line 156, in _path_isfile
  File "<frozen importlib._bootstrap_external>", line 148, in _path_is_mode_type
KeyboardInterrupt
[INFO] [rospy.core]: signal_shutdown [atexit]
[robot_state_publisher-2] killing on exit
[joint_state_publisher-1] killing on exit
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[state_estimator-14] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
hello-robot@stretch-re2-2003:~/catkin_ws/src/home_robot_hw/launch$ python -m home_robot.nodes.goto_controller
/usr/bin/python: Error while finding module specification for 'home_robot.nodes.goto_controller' (ModuleNotFoundError: No module named 'home_robot')

```

exhaustin, 2023-03-06:
1. Re-install home-robot via `pip install -e /path/to/home-robot/src/home_robot`
2. Rebuild catkin_ws and re-source setup.devel

cpaxton, 2023-05-24:
I think this is resolved?

closed 2023-05-24 by cpaxton