facebookresearch/home-robot #60: Collecting demonstrations as H5 file for learning

zephirefaith, 2023-02-22:
## 🚀 Feature
Scripts to enable collecting demonstration episodes as H5 files. The script would need to access all "relevant" data and write it to an H5 file as following:
- [ ] Insert H5 format here

The use-case is to collect demonstrations as H5s and directly use these H5 files using data_tools to  train our learning algorithms/introspect behavior.

## Central Idea

I will rely on the StretchEnv to create all the subscribers needed to collate data and return a dictionary at time _t_. The h5 collection script will be responsible for accessing this data at either (a) some pre-fixed frequency OR (b) a user-supervised trigger. This data will be collected as an episodic object and written to an H5 file. It's a straight-forward implementation however has a few key limitations. 

1. The H5Writer from data_tools takes a long-time to write images to H5 and is the frequency bottleneck (from what I remember it supports ~2-5FPS). If higher frequencies are needed, we can change this writing to match Robohive implementation: write images as pure np arrays compressed with H5 gzip compression. I have noticed this solution to support ~15FPS. 
2. If we write each demonstration to one giant H5 file, this file *will* get corrupted if mid-demonstration a system error pops up. This is because the H5Writer instance was latched onto the output file and active within the data collection loop. Any crash leads to improper closing of this object. To circumvent this issue we were dumping one H5 file per demonstration previously. However, if there is a strong reason to dump 1 H5 or fewer H5s, we can detach H5Writer instance from data collection loop by instead collecting information in a local variable. This variable can be written to H5 output file when the data collection loop exits normally. 

## Questions

There are two key parameters here: what is relevant data? what is the minimum frequency required for data collection?

### Relevant Data

Following is a starter list of basic robot data I can imagine being relevant:
- Base location
- Depth image
- RGB Image
- End-effector location
- Joint-values
I'd like opinion from all stakeholders to know if this aligns/if I am missing something.

### Minimum Frequency

For previous instance of data collector we were using a *trigger-based* data collection mechanism, i.e. only collect data when the user marks a _keyframe_. This meant we did not have any hard frequency requirements. I am not sure if that form of data collection still makes sense? I'd like to know if such requirements exist currently.

zephirefaith, 2023-02-22:
@cpaxton @exhaustin @theophilegervet @codekansas

SamNPowers, 2023-02-22:
Regarding your second issue, I have a fix for that here: https://github.com/fairinternal/data_tools/blob/spowers/clean_exit_h5/src/data_tools/writer.py

Also you might be interested in how I'm collecting data right now -- enables both keyframe collection and all frame collection (currently ros.spin'ing at 10Hz): https://github.com/cpaxton/home_robot/blob/spowers_xbox_ros_pbik/src/home_robot/envs/imitation/stretch_online_demo_env.py

By the way in https://github.com/cpaxton/home_robot/blob/spowers_xbox_ros_pbik/src/home_robot/envs/imitation/stretch_demo_base_env.py you can see how I'm loading trajectories -- you can record several episodes at once if you like (they'll be all in one h5) or you can stop between episodes and restart, and as long as you specify during environment loading directory where all of those h5s exist, all episodes will be treated the same. 

It'll also recurse if you specify a top level directory -- this is how I easily switch between groupings. E.g. if I want only one demo, I have a folder with that one demo. If I want to specify several demos for a particular task, I specify the higher level folder that contains all of them.

mhmukadam, 2023-02-23:
Tagging if you'll have any comments wrt lemda @luisenp, @KRobG, @rmurai0610

zephirefaith, 2023-02-23:
@SamNPowers Thanks for sharing your code! I like the changes in data_tools and will port those in. Still going through the `StretchOnlineDemoEnv`. It does things differently from what I had imagined.

Curious about how do you use StretchDemoBaseEnv...I see you are also creating an aggregated H5 file. Wondering why is this being done if you can choose any grouping based on path anyway? Do you provide this aggregated file to your DataLoader? Asking because DataLoader also has a recursive reading ability anyway.

zephirefaith, 2023-06-14:
Closed in above

closed 2023-06-14 by zephirefaith