facebookresearch/home-robot #64: Add Python type hinting

kalakris, 2023-02-23:
## 🚀 Feature
Add Python type hinting and automatic linting to our codebase

## Motivation
There are many functions where the arguments are return types are ambiguous, and may need some trial-and-error or careful reading of the code to figure out how to use them. Adding type hinting throughout the codebase (and enforcing it with linting) could help with readability and usability.

cpaxton, 2023-02-23:
I think this is a great idea, especially seeing the issues we had with @srama2512 - where a bunch of function calls had changed on hardware side.

Do you have recommendations for setting this up with linter? + @exhaustin

exhaustin, 2023-02-23:
Right now we have a lot of high-pri features to add / stuff to refactor. Legacy code gets modified and deleted every day so cleaning everything up and adding type hints for everything right now will result in some wasted efforts and a lot of merge conflicts.

What we can do right now is to make sure that whatever code we're adding / changing has type hints during PR reviews. The already integrated Mypy does static type checking, so it should help with that.
Once the codebase is stable we can try to do a full cleanup and add a linter that enforces all methods to be typed.

kalakris, 2023-02-23:
That sounds like a good plan. I do like the idea of adding type hints and docs as you go, particularly if you're refactoring code, since the API usage is most fresh on your mind at that point and should hopefully make it easier than doing it later.

I'll hold off on jumping in at this point.

cpaxton, 2023-05-24:
We're adding type hinting so I'll close this issue for now

closed 2023-05-24 by cpaxton