facebookresearch/home-robot #159: Can't download assets: scene data, robot, pretrained models etc

jiaming-robot-learning, 2023-05-02:
Hi,

I tried to install the repo according to the instructions, however I was unable to download the assets as instructed:

```
cd `HOME_ROBOT_ROOT/data/`
# Download the scenes
git clone https://huggingface.co/datasets/osmm/fpss --branch osmm
# Download the objects and metadata
git clone https://huggingface.co/datasets/osmm/objects
```
It asks for my huggingface user name and password. It seems this repo is private that we don't have access. 

Is it not public released yet?

cpaxton, 2023-05-24:
Yes sorry! Dataset will be publicly released in ~2 weeks hopefully

cpaxton, 2023-06-20:
https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md dataset download instructions are now up to date and the dataset should be public. Please re-open and let us know if you have any further issues!

closed 2023-06-20 by cpaxton

jiaming-robot-learning, 2023-06-21:
Hi Chris,

Thanks for the update! Yes the scene and object assets are now able to be downloaded.

However, the link for downloading the pre-trained skills as indicated in the README seems not working:
>https://huggingface.co/datasets/osmm/checkpoints

Could you please check that?