facebookresearch/home-robot #182: missing arguments in multiple places

yairshp, 2023-05-14:
## 🐛 Bug

I got multiple errors regarding missing arguments:

1. file - imagenav_agent.py, line - 33. method - constructor of Categorical2DSemanticMapModule. missing arguments - explored_radius, been_close_to_radius
2. file - imagenav_agent.py. line - 246. method - constructor of DiscretePlanner. missing arguments - step_size
3. file - imagenav_agent.py. line - 306. method - self.planner.plan. missing arguments - frontier_map

Also, the return value of of the self.planner.plan (see 3) is a tuple of size 4, while there are only 2 variables assigning the return value.

## Steps to Reproduce

Steps to reproduce the behavior:

```python projects/instanceimagenav/eval_episode_habitat.py```

## Expected behavior

arguments should be specified.

cpaxton, 2023-05-24:
@theophilegervet  @jacobkrantz - we should be able to fix this easily enough

cpaxton, 2023-05-25:
@yairshp could you tell me a bit about how you're using this? we haven't officially released the library yet so haven't been handling bug requests like this

yairshp, 2023-05-30:
@cpaxton 
Sure. I reproduced the steps mentioned [here](https://github.com/facebookresearch/home-robot/tree/main/projects/instanceimagenav#readme). when i tried to run the program using the command:
```python projects/instanceimagenav/eval_episode_habitat.py```
i got errors on missing arguments. i listed in my first message the exact places where it happened

cpaxton, 2023-08-01:
This should be addressed now

closed 2023-08-01 by cpaxton