facebookresearch/home-robot #218: Quality of life feature requests

ykarmesh, 2023-06-02:
## 🚀 Feature
Quality of life feature request:
1) Check for runstop before starting any movements
2) Allow people to use assertion to check if the robot is homed. 

## Motivation

When the agent is in runstop mode, the policy code has no idea about it right now. The policy code keeps sending commands to the robot and the robot doesnt move. This wastes time since the user now needs to kill the current code and restart things again after the runstop as been removed. 
Similarly, when we want to use the arm, it has to be homed first. If the agent isnt homed and there are some initial parts of the experiment that doesnt require manipulation, we could be wasting a large amount of time before realizing the mistake.

hello-binit, 2023-07-06:
Hey @cpaxton and @ykarmesh, thanks for bringing this up. We merged [a PR](https://github.com/hello-robot/stretch_ros/pull/105) today that gives you the [`/is_homed`](https://github.com/hello-robot/stretch_ros/tree/noetic/stretch_core#is_homed-std_msgsbool) and [`/is_runstopped`](https://github.com/hello-robot/stretch_ros/tree/noetic/stretch_core#is_runstopped-std_msgsbool) topics. I think the best place to add this would be to [`home_robot_hw.remote._create_pubs_subs()`](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot_hw/home_robot_hw/remote/ros.py#L235-L237).

cpaxton, 2023-08-15:
Finally fixed here https://github.com/facebookresearch/home-robot/pull/338

closed 2023-08-16 by cpaxton