facebookresearch/home-robot #238: Unable to install habit-sim by building from source on Ubuntu 22.04

suyashkumar2409, 2023-06-20:
## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1. Following the installation steps [here](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot_sim/README.md) on an Ubuntu 22.04

## Expected behavior

The step
`python setup.py install --bullet`
should succeed.

Instead, the script crashes my PC halfway at a very specific step at around 43%.

## Additional context

https://github.com/facebookresearch/home-robot/assets/10911412/3215b256-74c9-421d-bb48-0874f7f82eb5

suyashkumar2409, 2023-06-20:
It seems that this installation is causing an OOM for my device, and a serial installation succeeds. Raised a PR to update documentation for the same.

https://github.com/facebookresearch/home-robot/pull/239

cpaxton, 2023-06-20:
Thanks for figuring this out! Will review and hopefully merge this today.

cpaxton, 2023-06-20:
@suyashkumar2409  - we just updated install instructions so you no longer need to build from source. Hopefully it solves the issue?

cpaxton, 2023-06-29:
Closing as hopefully resolved

closed 2023-06-29 by cpaxton