facebookresearch/home-robot #243: Can't run eval_dataset: creating robot failed

jiaming-robot-learning, 2023-06-21:
## 🐛 Bug

The ovmm evaluation script fails to run after a fresh installation following the instructions.

## Steps to Reproduce

Steps to reproduce the behavior:

Simply run 
```bash
python projects/habitat_ovmm/eval_dataset.py
```

The error message is:
-----
pybullet build time: May 20 2022 19:45:31
2023-06-21 11:26:43,692 Initializing dataset OVMMDataset-v0
2023-06-21 11:26:44,468 initializing sim OVMMSim-v0
[11:26:45:367752]:[Core] ManagedContainerBase.h(329)::checkExistsWithMessage : <Lighting Layout>::getObjectByHandle : Unknown Lighting Layout managed object handle :  . Aborting
2023-06-21 11:26:46,012 Initializing task OVMMNavToObjTask-v0
/home/users/jiaming/ovmm/ovmm_env/lib/python3.9/site-packages/gym/spaces/box.py:84: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
Running eval on [1200] episodes
[11:26:46:852964]:[Core] ManagedContainerBase.h(205)::getObjectHandleByID : Unknown ArticulatedObject managed object ID: -1 . Aborting
[11:26:46:852999]:[Core] ManagedContainerBase.h(329)::checkExistsWithMessage : <ArticulatedObject>::getObjectCopyByID : Unknown ArticulatedObject managed object handle :  . Aborting
Process ForkServerProcess-1:
Traceback (most recent call last):
  File "/home/users/jiaming/ovmm/ovmm_env/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/users/jiaming/ovmm/ovmm_env/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/users/jiaming/ovmm/ovmm_env/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/vector_env.py", line 281, in _worker_env
    result = result_or_fn(**function_args)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/gym_env_episode_count_wrapper.py", line 61, in reset
    return self.env.reset(**kwargs)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/gym_env_obs_dict_wrapper.py", line 36, in reset
    return self.env.reset(**kwargs)
  File "/home/users/jiaming/ovmm/home-robot/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py", line 107, in reset
    habitat_obs = self.habitat_env.reset()
  File "/home/users/jiaming/ovmm/ovmm_env/lib/python3.9/site-packages/gym/core.py", line 283, in reset
    return self.env.reset(**kwargs)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/utils/gym_adapter.py", line 300, in reset
    obs = self._env.reset()
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 54, in reset
    observations = super().reset()
  File "/home/users/jiaming/ovmm/ovmm_env/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 422, in reset
    return self._env.reset()
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 260, in reset
    self.reconfigure(self._config)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 342, in reconfigure
    self._sim.reconfigure(self._config.simulator, self.current_episode)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 260, in reconfigure
    self.robots_mgr.reconfigure(new_scene)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/robot_manager.py", line 102, in reconfigure
    robot_data.robot.reconfigure()
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/robots/mobile_manipulator.py", line 143, in reconfigure
    Manipulator.reconfigure(self)
  File "/home/users/jiaming/ovmm/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/robots/manipulator.py", line 82, in reconfigure
    self.sim_obj.auto_clamp_joint_limits = True
AttributeError: 'NoneType' object has no attribute 'auto_clamp_joint_limits'
....
---------

## Expected behavior

The evaluation should run successfully.

## Additional context

I locate the problem being the following line in "home-robot/src/third_party/habitat-lab/habitat-lab/habitat/robots/manipulator.py"
```python
def reconfigure(self) -> None:
        """Instantiates the robot the scene. Loads the URDF, sets initial state of parameters, joints, motors, etc..."""
        # TODO: The current implementation requires users to define all the components of the robot in a single URDF.
        # The future will support loading multiple URDF files.
        if self.sim_obj is None or not self.sim_obj.is_alive:
            ao_mgr = self._sim.get_articulated_object_manager()
            self.sim_obj = ao_mgr.add_articulated_object_from_urdf(  # <<< THIS FAILS AND RETURNS NONE
                self.urdf_path, fixed_base=self._fixed_base
            )
```
I confirm the urdf file exists and urdf_path correctly points to the file. I also created a simple URDF and called _ao_mgr.add_articulated_object_from_urdf_. The fails again and the error message is the same as follows:

------
[12:15:29:130992]:[Core] ManagedContainerBase.h(205)::getObjectHandleByID : Unknown ArticulatedObject managed object ID: -1 . Aborting
[12:15:29:131056]:[Core] ManagedContainerBase.h(329)::checkExistsWithMessage : <ArticulatedObject>::getObjectCopyByID : Unknown ArticulatedObject managed object handle :  . Aborting
-------

jiaming-robot-learning, 2023-06-21:
Solved by re-build habitat-sim with the bullet option.

yvsriram, 2023-06-21:
Thanks for figuring this out! We are updating the instructions [here](https://github.com/facebookresearch/home-robot/pull/244) to install habtiat-sim withbullet always.