facebookresearch/home-robot #249: ValueError: Cannot load file containing pickled data when allow_pickle=False

lchen1019, 2023-06-22:
I have successfully completed the above steps, but when running eval_dataset.py, I encountered the following error. I tried adding allow_pickle=True, but got new errors.

(home-robot) chenlin@sg1:/data/chenlin/home-robot$ python projects/habitat_ovmm/eval_dataset.py
pybullet build time: May 20 2022 19:45:31
/data/chenlin/anaconda3/envs/home-robot/lib/python3.9/site-packages/torchvision/io/image.py:13: UserWarning: Failed to load image Python extension: '/data/chenlin/anaconda3/envs/home-robot/lib/python3.9/site-packages/torchvision/image.so: undefined symbol: _ZNK3c107SymBool10guard_boolEPKcl'If you don't plan on using image functionality from `torchvision.io`, you can ignore this warning. Otherwise, there might be something wrong with your environment. Did you have `libjpeg` or `libpng` installed before building `torchvision` from source?
  warn(
Arguments:
{
    "habitat_config_path": "ovmm/ovmm_eval.yaml",
    "baseline_config_path": "projects/habitat_ovmm/configs/agent/hssd_eval.yaml",
    "opts": []
}
----------------------------------------------------------------------------------------------------
Configs:

----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/data/chenlin/home-robot/projects/habitat_ovmm/eval_dataset.py", line 254, in <module>
    evaluator.eval(
  File "/data/chenlin/home-robot/projects/habitat_ovmm/eval_dataset.py", line 81, in eval
    self._init_envs(
  File "/data/chenlin/home-robot/src/third_party/habitat-lab/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py", line 215, in _init_envs
    self.envs = construct_envs(
  File "/data/chenlin/home-robot/src/third_party/habitat-lab/habitat-baselines/habitat_baselines/common/construct_vector_env.py", line 42, in construct_envs
    scenes = dataset.get_scenes_to_load(config.habitat.dataset)
  File "/data/chenlin/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 64, in get_scenes_to_load
    dataset = cls(cfg)
  File "/data/chenlin/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/datasets/ovmm/ovmm_dataset.py", line 119, in __init__
    self.viewpoints_matrix = np.load(
  File "/data/chenlin/anaconda3/envs/home-robot/lib/python3.9/site-packages/numpy/lib/npyio.py", line 438, in load
    raise ValueError("Cannot load file containing pickled data "
ValueError: Cannot load file containing pickled data when allow_pickle=False

yvsriram, 2023-06-22:
Hi, thanks for reporting the issue!

This means data did not get downloaded for you when using `git submodule update --init ` commands. Probably this only downloaded pointers from you. To verify if that is the case, can you please try running size of the file `data/datasets/ovmm/train/viewpoints.npy`.

You can install it via git-lfs:
1. Install `pip install git-lfs`
2. cd to each data directory (eg. `cd data/hssd-hab` and run `git lfs pull`. 

Let us know if this doesn't work for you.

prasoonvarshney, 2023-06-22:
Had the exact same problem and the recommended solution solve the issue, thanks!

SiyuanHuang95, 2023-06-26:
hi guys, @yvsriram @prasoonvarshney, 

do you install this repo in one local workstation or in the slurm.

I can install it in one local workstation, but since the slurm I cannot have the root rights, it failed many times...

yvsriram, 2023-06-29:
@SiyuanHuang95 I was able to install this in our university cluster without having root privileges.

SiyuanHuang95, 2023-06-29:
Hi, @yvsriram, I have figured out what the installation needs:

1. it needs the Nvidia-driver full version, not the driverless one, as stated here: https://github.com/facebookresearch/habitat-sim/issues/1671

2. it needs the update of GXXLIB, at least the right to do some ln -s operation.

closed 2023-06-29 by yvsriram