facebookresearch/home-robot #251: Install home_robot_sim failed and confusion with step 8

SiyuanHuang95, 2023-06-23:
## 🐛 Bug

First of all, very cool project! Thanks for sharing!

when trying to install home_robot_sim following [HERE](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot_sim/README.md), the error raised: 
```
ERROR: Could not find a version that satisfies the requirement home-robot (from home-robot-sim)
ERROR: No matching distribution found for home-robot
```

Also, I am quite confused about step 8 [Readme](https://github.com/facebookresearch/home-robot#8-run-open-vocabulary-mobile-manipulation-on-stretch), both server seems to involve the usage of ROS, but it says in [Preliminary](https://github.com/facebookresearch/home-robot#preliminary), no need for ROS when using without hardware. Could you also provide the demo case for the usage without ROS?

Also, I found in Step 7, **conda activate contact_graspnet_env**, but where it come from?

## Steps to Reproduce

Steps to reproduce the behavior:

## Expected behavior

Would it be possible to use Docker, or you guys could provide a shell script consisting of all these steps?

## Additional context

yvsriram, 2023-06-23:
Hi Siyuan, 

Thanks for creating the issue!

> when trying to install home_robot_sim following [HERE](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot_sim/README.md), the error raised:

 Did you install `home-robot` with `python -m pip install -e src/home_robot` in Step 2

> Also, I am quite confused about step 8 [Readme](https://github.com/facebookresearch/home-robot#8-run-open-vocabulary-mobile-manipulation-on-stretch), both server seems to involve the usage of ROS, [...]

Step 8 is not relevant when using without hardware. We will make this clearer.

> Also, I found in Step 7, conda activate contact_graspnet_env, but where it come from?

This is in step 8, right? 

> Would it be possible to use Docker, or you guys could provide a shell script consisting of all these steps?

Great idea! We will add one of these to make the installation easier.

SiyuanHuang95, 2023-06-23:
Hi,

Great thanks for your speedy reply! Appreciate!

Yes, I am jumping over step 2, will try it!

If we are using the version without ROS, how do we test the installation's success with a quick demo?

And yes, contact_graspnet_env is in step 8.

yvsriram, 2023-06-23:
Great that it got fixed.

> If we are using the version without ROS, how do we test the installation's success with a quick demo? Should we follow the setting [Here](https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md)?

Yes please follow the [Habitat OVMM README](https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md)

closed 2023-06-23 by SiyuanHuang95