facebookresearch/home-robot #252: Any possibility to replace the usage of sophuspy

SiyuanHuang95, 2023-06-25:
## 🐛 Bug

In the cluster, the user generally does not have the root right, which makes the installation of the required packs quite difficult. 

To install the sophuspy, the user needs to install pybind previously and link many static libraries, which would be difficult for the guys without root.

## Steps to Reproduce

Steps to reproduce the behavior:

## Expected behavior

## Additional context

SiyuanHuang95, 2023-06-25:
Also, one very interesting thing is that, it seems that you did not use sophuspy actually in the source code.

yvsriram, 2023-06-28:
Yes, you are right. It is not used anywhere and will be removed from the list of requirements.

SiyuanHuang95, 2023-06-29:
Cool, thanks!

yvsriram, 2023-07-02:
It seems like `sophuspy` is actually imported in files under `src/home_robot` (as `import sophus`).  Also, these functions are being used in `discrete_planner`. So the dependency cannot be trivially removed.

yvsriram, 2023-07-02:
Are you still facing issues installing this dependency?

SiyuanHuang95, 2023-07-02:
To be honest, I did not successfully install the home-robot on the slurm.

But, I somehow make it in one workstation where I have sudo rights.

cpaxton, 2023-07-03:
What issues are you still having?

SiyuanHuang95, 2023-07-04:
@cpaxton sophuspy needs pybind, pybind needs make and install, which needs the sudo rights.

closed 2023-07-25 by SiyuanHuang95