facebookresearch/home-robot #257: Documentation Request for Generating Expert Trajectories

prasoonvarshney, 2023-06-29:
Hi

Is there a way we get access to expert trajectories for the training data in case we want to do imitation learning?

Also, what's the best way to generate longer horizon tasks like tidying up a house which might include open/close actions and multiple pick/place actions?

Thank you!

cpaxton, 2023-06-29:
We are currently just doing RL training, we're looking into generating some expert trajectories though based on these policies

cpaxton, 2023-06-29:
This is something we're looking into, but you could also use our heuristic policy to generate expert data!

prasoonvarshney, 2023-06-29:
Re: you could use our heuristic policy to generate expert data - 

I tried the heuristic policy for the NAV_TO_OBJ and NAV_TO_REC sub-tasks in OVMM episodes, and from the resulting videos, the Stretch Robot seems to be exploring its environment, progressively building a semantic map, to find the target object. 

Is there a way to enable exposing the full semantic scene map at the start of the episode, in order to do oracle navigation (A* planning given start and target location co-ordinates), so that we can generate expert demonstration videos where the agent is assumed to have prior knowledge of where each object is kept in the environment?

closed 2023-09-11 by cpaxton