facebookresearch/home-robot #258: Documentation Request for Episode Format

prasoonvarshney, 2023-06-29:
Hi, what are the semantic meanings of the fields in the episode JSON files at the [HuggingFace OVMM dataset page](https://huggingface.co/datasets/ai-habitat/OVMM_dataset/tree/main/train)?

For instance, target goals seem to be defined as a list of 4-tuples, and I wasn’t able to understand if they refer to xyz coordinates and a single angle w along the z-axis, or something else.
```
"targets": {
        "Olive_Kids_Dinosaur_Land_Lunch_Box_:0000": [
            [
                0.70621,
                0,
                -0.708,
                -15.78765
            ],
            [
                0,
                1,
                0,
                0.87444
            ],
            [
                0.708,
                0,
                0.70621,
                0.66923
            ],
            [
                0,
                0,
                0,
                1
            ]
        ]
    },
```

yvsriram, 2023-06-29:
Thanks for raising this!

This `targets` field is not used in Ovmm task. It is used for specifying a target goal (target transformation for the object) in the [GeoGoal rearrange task](https://aihabitat.org/challenge/2022_rearrange/). 

Added an initial description of the OVMM task-relevant fields [here](https://huggingface.co/datasets/ai-habitat/OVMM_dataset/discussions/1/files#d2h-611991). This will be visible on the huggingface dataset page soon.

prasoonvarshney, 2023-06-29:
Sounds good, thanks for the updated documentation @yvsriram!

As a follow-up, and apologies about digressing a bit from the OVMM challenge - 

Is there a recommended way to use episode generation code in the `habitat-lab` repo to generate longer horizon tasks that might involve open/close actions or a sequential sequence of pick/place actions, like decluttering/tidying a living room? 
(I realize this would be a departure from the specification for OVMMDatasetV0 class, but in general looking for how to perform more complex task sequences.)

yvsriram, 2023-06-30:
For multiple pick/place interactions: You can look at the [configs](https://github.com/facebookresearch/habitat-lab/tree/ovmm/habitat-lab/habitat/datasets/rearrange/configs/hab) from [hab2](https://arxiv.org/pdf/2106.14405.pdf) tasks. You will have to modify this [script](https://github.com/facebookresearch/habitat-lab/blob/ovmm/habitat-lab/habitat/datasets/rearrange/modify_episodes_for_object_rearrange.py) for retrieving categories of multiple goals.

yvsriram, 2023-06-30:
For open/close actions: Currently OVMM uses HSSD scenes, which do not have articulated receptacles. So open/close actions cannot be performed on them. 

But you can create episodes using ReplicaCAD scenes: Look at `rearrange.yaml` [here](https://github.com/facebookresearch/habitat-lab/tree/ovmm/habitat-lab/habitat/datasets/rearrange/configs/hab). `rearrange_easy.yaml` is the one which does not have open/close interactions.

cpaxton, 2023-06-30:
I think in the fall we're hoping to add articulations, but yeah not available yet.

We can also continue this discussion on Slack @prasoonvarshney

cpaxton, 2023-08-04:
So we're probably not going to add docs to the episode setup as it currently is, will hopefully have a better version soon. Closing the issue