facebookresearch/home-robot #259: Loading a non-vectorized OVMM Environment is very slow

trangml-mit, 2023-06-29:
I am beginning to work with the `HabitatOpenVocabManipEnv` and wanted to load in a single environment instance to do evaluation on. However, I noticed that in `projects/habitat_ovmm` there is no example of a single `eval_episode.py`, only `eval_dataset.py`, which seems prohibitive to rapid testing.

When experimenting with running a single environment instance with the following code
```
config_path = "ovmm/rl_skill.yaml"
config, config_str = get_config(config_path)
OmegaConf.set_readonly(config, False)

config.habitat_baselines.num_environments = 1
OmegaConf.set_struct(config.habitat_baselines, False)
config.habitat_baselines.print_images = 1
config.habitat.dataset.split = "val"
config.habitat_baselines.exp_name = "debug"

OmegaConf.set_readonly(config, True)
baseline_config = OmegaConf.load(
    "projects/habitat_ovmm/configs/agent/hssd_eval.yaml"
)
config = DictConfig({**config, **baseline_config})

agent = OpenVocabManipAgent(config=config)
env = HabitatOpenVocabManipEnv(Env(config=config.habitat), config=config)
```

The code stalls on the loading of `HabitatOpenVocabManipEnv`, and the last output I see is 

```
2023-06-29 10:12:36,567 Initializing dataset OVMMDataset-v0
2023-06-29 10:12:37,610 initializing sim OVMMSim-v0
```

Even after leaving the code to run for multiple hours the OVMMSim does not seem to fully load.

Is there an alternative to loading in a single instance of `HabitatOpenVocabManipEnv` that does not require loading the entire OVMMDataset, or is there a way to make a OVMM dataset smaller for testing?

yvsriram, 2023-06-29:
> The code stalls on the loading of HabitatOpenVocabManipEnv

Are you sure that the code is stalling for you? Can you check if the outputs under `datadump/` directory are empty for you.

> or is there a way to make a OVMM dataset smaller for testing?

Yes, you can evaluate on a smaller dataset. Use `habitat.dataset.episode_ids=[0]` to specify a list of episode ids (see [this](https://github.com/facebookresearch/home-robot/tree/main/projects/habitat_ovmm#running-evaluations)). You can add `habitat.dataset.episode_indices_range=[0,10]` to run evaluations on a contiguous episode set.

trangml-mit, 2023-06-29:
Interestingly, I think it might be a problem with initializing the environments in my Jupyter notebook. 

Doing some tests with the ObjectNav environment, the first time I try to load an env using 
`env = HabitatObjectNavEnv(Env(config=config), config=config)`
it stalls and does nothing after printing
```
2023-06-29 11:42:06,453 Initializing dataset ObjectNav-v1
2023-06-29 11:42:10,068 initializing sim Sim-v0
```

If I wait a while, then interrupt the kernel running the line, once the kernel stops running the cell, I get the following output
```
[11:42:10:082565]:[Metadata] SceneDatasetAttributesManager.cpp(310)::readDatasetJSONCell : "stages.default_attributes" set in Attributes Manager from JSON.
[11:42:10:082590]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d_v0.2/example/00861-GLAQ4DNUx5U/*.basis.glb
[11:42:10:082622]:[Metadata] AttributesManagerBase.h(352)::buildAttrSrcPathsFromJSONAndLoad : Glob path result for ./data/scene_datasets/hm3d_v0.2/minival/00800-TEEsavR23oF/*.basis.glb : ./data/scene_datasets/hm3d_v0.2/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb
[11:42:10:082707]:[Metadata] AttributesManagerBase.h(352)::buildAttrSrcPathsFromJSONAndLoad : Glob path result for ./data/scene_datasets/hm3d_v0.2/minival/00802-wcojb4TFT35/*.basis.glb : ./data/scene_datasets/hm3d_v0.2/minival/00802-wcojb4TFT35/wcojb4TFT35.basis.glb
[11:42:10:082776]:[Metadata] AttributesManagerBase.h(352)::buildAttrSrcPathsFromJSONAndLoad : Glob path result for ./data/scene_datasets/hm3d_v0.2/minival/00803-k1cupFYWXJ6/*.basis.glb : ./data/scene_datasets/hm3d_v0.2/minival/00803-k1cupFYWXJ6/k1cupFYWXJ6.basis.glb
[11:42:10:082848]:[Metadata] AttributesManagerBase.h(352)::buildAttrSrcPathsFromJSONAndLoad : Glob path result for ./data/scene_datasets/hm3d_v0.2/minival/00808-y9hTuugGdiq/*.basis.glb : ./data/scene_datasets/hm3d_v0.2/minival/00808-y9hTuugGdiq/y9hTuugGdiq.basis.glb
[11:42:10:082907]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d_v0.2/test/00900-XRJ2muKkwAV/*.basis.glb
[11:42:10:082918]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d_v0.2/test/00901-aJoC8Qw6xQ5/*.basis.glb
[11:42:10:082929]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d_v0.2/test/00904-S9CUp5RsFY9/*.basis.glb
[11:42:10:082939]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d_v0.2/test/00905-9v5HzFAB1cm/*.basis.glb
...(truncated for brevity)
```
Then, if I run the cell again immediately after, it successfully loads the task and the environment.
```
2023-06-29 11:46:38,497 Initializing dataset ObjectNav-v1
2023-06-29 11:46:41,841 initializing sim Sim-v0
2023-06-29 11:46:44,351 Initializing task ObjectNav-v1
```

So I'm not really sure why, but it seems like when running in a Jupyter notebook something in loading the scenes gets stuck, but after a manual stop and restart it no longer gets stuck. 

I did notice that running the same code regularly via python worked fine, just Jupyter seemed broken.

cpaxton, 2023-06-29:
Interesting. Is this a problem with glob? Is it just endlessly printing that one line?

Regardless, if you're working on an Ipython notebook, i think that would be an amazing contribution to our codebase to lower barrier of entry - could you open a PR, and maybe we can help try to fix it?

cpaxton, 2023-06-29:
Can i ask what happens if you just run something like 
```
os.listdir('./data/scene_datasets/hm3d_v0.2/test/00905-9v5HzFAB1cm/')
os.listdir('./data/scene_datasets/hm3d_v0.2/test/)
```
Those paths exist the first time through right?

trangml-mit, 2023-06-29:
I haven't downloaded the test set for hm3d_v0.2, all I have is the val and minival set. So glob is correct in not finding it, but I'm confused on why it's looking for it in the first place and getting stuck on it when running as a python module doesn't. The actual output after stopping the cell is massive, with around 300 lines of output for the success loading and then 300 lines of the Glob complaints. The Glob complaints all seem to be about missing train and test files.

Is it required to download the entire hm3d dataset, and not just a subset?

I can work on submitting a PR with the notebook.

cpaxton, 2023-06-30:
@trangml-mit we need you to accept the CLA so I can review + merge this, fyi

trangml-mit, 2023-06-30:
I'm working on this as part of my work, so I unfortunately am stuck waiting on approval to accept the CLA. I don't anticipate this being resolved quickly over the holiday weekend so if anyone else is able to make a PR that may be a better solution. The actual notebook is very simple, so I don't think it'd be hard.

cpaxton, 2023-06-30:
Got it. We can take a look at the notebook without the CLA, no worries.