facebookresearch/home-robot #313: ModuleNotFoundError: No module named 'habitat.utils.gym_definitions'

xtli12, 2023-07-31:
## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:
When I followed the [Demo Setup](https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md#demo-setup) in Habitat OVMM readme like:
`python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml`

the error came out like:
```
pybullet build time: May 20 2022 19:45:31
Traceback (most recent call last):
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/eval_baselines_agent.py", line 10, in <module>
    from evaluator import OVMMEvaluator
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/evaluator.py", line 19, in <module>
    from utils.env_utils import create_ovmm_env_fn
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/utils/env_utils.py", line 11, in <module>
    from habitat.utils.gym_definitions import _get_env_name
ModuleNotFoundError: No module named 'habitat.utils.gym_definitions'
```

I had already installed `habitat-lab`, `habitat-baselines` and `pytorch3d`, but the error still exits

cpaxton, 2023-07-31:
Did you install our versions of these libraries, or did you install from something else?

cpaxton, 2023-07-31:
Can you please check the versions of all 3 libraries and share them here?

xtli12, 2023-08-01:
The version of  all 3 libraries are
 
[image]
[image]
Because I'm in China, I can't install these libraries on my GPU server directly via github, so I download these packages on my  computer first, then upload these packages to my GPU server and install them.

cpaxton, 2023-08-01:
Are you sure they're checked out from the same commits in our codebase? I would check in each branch to make sure. You can check on your computer, make sure to run `git submodule update --init --recursive`, and copy them out.

I'm guessing this is where your issue is

yvsriram, 2023-08-01:
Yes, please do as @cpaxton suggested. The `habitat` version should be 0.2.3 on the right `habitat-lab` commit.

xtli12, 2023-08-02:
When I attempted to run the command `git submodule update --init --recursive src/third_party/habitat-lab`, there was no output or response. 
then I try to run `git pull https://github.com/facebookresearch/habitat-lab src/third_party/habitat-lab`, it turns out that my GPU server was unable to access `github`, so I resorted to an alternative method: I download the `habitat-lab-main` on my local computer via 'https://github.com/facebookresearch/habitat-lab/', then upload the  `habitat-lab-main` repository to my GPU server and install it, the information is as follows:
```
(home1) lxt21@ubuntu:~/home-robot-main$ git submodule update --init --recursive src/third_party/habitat-lab
(home1) lxt21@ubuntu:~/home-robot-main$ git pull https://github.com/facebookresearch/habitat-lab src/third_party/habitat-lab 
fatal: unable to access 'https://github.com/facebookresearch/habitat-lab/': Failed to connect to 127.0.0.1 port 1080: Connection refused
(home1) lxt21@ubuntu:~/home-robot-main$ python -m pip install -e src/third_party/habitat-lab-main/habitat-lab
```
I also installed the `habitat-baselines` in the same way, but the error still exists when I try to setup demo:
```
(home1) lxt21@ubuntu:~/home-robot-main$ python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml
pybullet build time: May 20 2022 19:45:31
Traceback (most recent call last):
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/eval_baselines_agent.py", line 10, in <module>
    from evaluator import OVMMEvaluator
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/evaluator.py", line 19, in <module>
    from utils.env_utils import create_ovmm_env_fn
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/utils/env_utils.py", line 11, in <module>
    from habitat.utils.gym_definitions import _get_env_name
ModuleNotFoundError: No module named 'habitat.utils.gym_definitions'
```

yvsriram, 2023-08-02:
Please download the `ovmm` branch instead of `main` branch: use `git clone https://github.com/facebookresearch/habitat-lab.git --branch ovmm`

xtli12, 2023-08-02:
Sorry, do you mean install the `habitat-lab` under the path `~/home-robot/data/datasets/ovmm`?
When I run `git clone https://github.com/facebookresearch/habitat-lab.git --branch ovmm`, the information is follows:
```
(home1) lxt21@ubuntu:~/home-robot-main$ git clone https://github.com/facebookresearch/habitat-lab.git --branch ovmm
fatal: destination path 'habitat-lab' already exists and is not an empty directory.
```
Is there any other way to install the `habitat-lab` without using `git` due to the limitations in accessing github from my GPU server.

yvsriram, 2023-08-02:
> Sorry, do you mean install the habitat-lab under the path ~/home-robot/data/datasets/ovmm?

No. I mean download the `ovmm` branch of `habitat-lab` on your local machine (you can also `git checkout ovmm`) and then move it to your gpu server. 

> Is there any other way to install the habitat-lab without using git due to the limitations in accessing github from my GPU server.

Can you connect to any external website from your gpu server? Maybe there is a connectivity issue.

xtli12, 2023-08-02:
Sorry, I haven't found any folder or file named `ovmm` in `habitat-lab`(https://github.com/facebookresearch/habitat-lab).

> Can you connect to any external website from your gpu server? Maybe there is a connectivity issue. ] 

The connection is ok, but my gpu server can't access to some website like Google, Youtube, GitHub, and so on in China.

yvsriram, 2023-08-02:
I was not asking you to download a specific folder or file. Can you try the following on your local machine?:

`git clone https://github.com/facebookresearch/habitat-lab.git --branch ovmm` at a path where `habitat-lab` doesn't already exist. This will download the `ovmm` branch (and not `main` branch), which contains code specific to the OVMM task. You can then use this version of `habitat-lab` code under `src/third_party`.

xtli12, 2023-08-03:
I have solved the issues with `habitat-lab`, thanks to your kind assistance!

However, I have encountered another problem. When I try to run the following command:
`python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml`
An error message is displayed, indicating the following:
```
[08:48:01:318874]:[Gfx] WindowlessContext.h(17)::~WindowlessContext : Deconstructing WindowlessContext
Traceback (most recent call last):
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/eval_baselines_agent.py", line 97, in <module>
    metrics = evaluator.evaluate(
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/evaluator.py", line 458, in evaluate
    self._env = create_ovmm_env_fn(self.config)
  File "/data/home-gxu/lxt21/home-robot-main/projects/habitat_ovmm/utils/env_utils.py", line 34, in create_ovmm_env_fn
    habitat_env = env_class(config=habitat_config, dataset=dataset)
  File "/data/home-gxu/lxt21/home-robot-main/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 117, in __init__
    base_env = RLTaskEnv(config=config, dataset=dataset)
  File "/data/home-gxu/lxt21/home-robot-main/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 43, in __init__
    super().__init__(config, dataset)
  File "/data/home-gxu/lxt21/home-robot-main/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 382, in __init__
    self._env = Env(config, dataset)
  File "/data/home-gxu/lxt21/home-robot-main/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 114, in __init__
    self._sim = make_sim(
  File "/data/home-gxu/lxt21/home-robot-main/src/third_party/habitat-lab/habitat-lab/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/data/home-gxu/lxt21/home-robot-main/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 82, in __init__
    super().__init__(config)
  File "/data/home-gxu/lxt21/home-robot-main/src/third_party/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 289, in __init__
    super().__init__(self.sim_config)
  File "<attrs generated init habitat_sim.simulator.Simulator>", line 11, in __init__
  File "/data/home-gxu/lxt21/mambaforge-pypy3/envs/home1/lib/python3.9/site-packages/habitat_sim-0.2.4-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 118, in __attrs_post_init__
    self.__set_from_config(self.config)
  File "/data/home-gxu/lxt21/mambaforge-pypy3/envs/home1/lib/python3.9/site-packages/habitat_sim-0.2.4-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 227, in __set_from_config
    self._config_backend(config)
  File "/data/home-gxu/lxt21/mambaforge-pypy3/envs/home1/lib/python3.9/site-packages/habitat_sim-0.2.4-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 200, in _config_backend
    super().__init__(config.sim_cfg, config.metadata_mediator)
AssertionError: ESP_CHECK failed: Missing (at least) one of scene dataset attributes, stage attributes, or dataset scene attributes
 for scene 'data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json'.  Likely an invalid scene name.
```
Does this issue relate to the data? I downloaded those data using the script to my local machine and then uploaded it to my GPU server:
```
git clone https://huggingface.co/datasets/hssd/hssd-hab
git clone https://huggingface.co/datasets/ai-habitat/OVMM_dataset
git clone https://huggingface.co/datasets/ai-habitat/OVMM_objects
```
but when I downloaded the `OVMM_objects`, I encountered some `filename-related` issues:
```
error: unable to create file train_val/google_scanned/assets/objects/Eat_to_Live_The_Amazing_NutrientRich_Program_for_Fast_and_Sustained_Weight_Loss_Revised_Edition_Book/meshes/Eat_to_Live_The_Amazing_NutrientRich_Program_for_Fast_and_Sustained_Weight_Loss_Revised_Editio
_The_Amazing_NutrientRich_Program_for_Fast_and_Sustained_Weight_Loss_Revised_Edi
tion_Book/meshes/Eat_to_Live_The_Amazing_NutrientRich_Program_for_Fast_and_Sustat_Canon_CLI_8CMY_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count/meshes/Office_Depot_Canon_CLI_8CMY_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count.object_conf
ined_Weight_Loss_Revised_Edition_Book.object_config.json: Filename too long
error: unable to create file train_val/google_scanned/assets/objects/Office_Depot_HP_2_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count/meshes/Office_Depot_HP_2_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count.object_config.json: Filename to
t_Canon_CLI_8CMY_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count
/meshes/Office_Depot_Canon_CLI_8CMY_Remanufactured_Ink_Cartridges_Color_Cyan_Magt_HP_564XL_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count/meshes/Office_Depot_HP_564XL_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count.object_config.json: Fil
enta_Yellow_3_count.object_config.json: Filename too long
error: unable to create file train_val/google_scanned/assets/objects/Office_Depot_HP_920XL_920_High_Yield_Black_and_Standard_CMY_Color_Ink_Cartridges/meshes/Office_Depot_HP_920XL_920_High_Yield_Black_and_Standard_CMY_Color_Ink_Cartridges.object_config.json: Filename too
t_HP_2_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count/meshes/Of
fice_Depot_HP_2_Remanufactured_Ink_Cartridges_Color_Cyan_Magenta_Yellow_3_count.
[... cut ...]
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```
I have tried using `git lfs clone`, but the error of `Filename too long` still exits. Could this issue be related to or the cause of the error in the Demo setup? 
```
AssertionError: ESP_CHECK failed: Missing (at least) one of scene dataset attributes, stage attributes, or dataset scene attributes 
for scene 'data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json'.  Likely an invalid scene name.
```

yvsriram, 2023-08-03:
> AssertionError: ESP_CHECK failed: Missing (at least) one of scene dataset attributes, stage attributes, or dataset scene attributes
 for scene 'data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json'.  Likely an invalid scene name.

For the first error, can you please check if the file `data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json` exists and is not empty?

> Does this issue relate to the data? I downloaded those data using the script to my local machine and then uploaded it to my GPU server:

Yes, can you please report the total size of the downloaded `hssd-hab` and `objects` folders.

> but when I downloaded the OVMM_objects, I encountered some filename-related issues:

Were you downloading the objects on Windows? Regardless, can you try the solutions [here]( https://stackoverflow.com/questions/22575662/filename-too-long-in-git-for-windows)

> I have tried using git lfs clone, but the error of Filename too long still exits. Could this issue be related to or the cause of the error in the Demo setup?

No, both errors seem unrelated to me - two separate problems that need to be solved.

xtli12, 2023-08-03:
Thanks for your immediate response!

>Were you downloading the objects on Windows? Regardless, can you try the solutions [here](https://stackoverflow.com/questions/22575662/filename-too-long-in-git-for-windows)

The error of Filename too long had been solved, thanks a lot! 
I have re-downloaded those data, but the error:
```
AssertionError: ESP_CHECK failed: Missing (at least) one of scene dataset attributes, stage attributes, or dataset scene attributes 
for scene 'data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json'.  Likely an invalid scene name.
```
still exits.
>For the first error, can you please check if the file ` data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance. json ` exists and is not empty?

The file `104348361_171513414.scene_instance. json` exit, but it is under the folder `data/hssd-hab/scenes/` not `data/hssd-hab/scenes-uncluttered/`, and the folder `data/hssd-hab/scenes-uncluttered/` doesn't exist in `hssd-hab`.

>Yes, can you please report the total size of the downloaded hssd-hab and objects folders?

The size of the downloaded `hssd-hab` and `objects` folders are 16.8GB and 1.66GB, respectively.

yvsriram, 2023-08-03:
You might have downloaded the wrong version of assets. If you are using `git clone` to download `hssd-hab, try adding  `--branch ovmm` to your command. This would have been taken care of if you download it via git submodules.

cpaxton, 2023-08-03:
I would really recommend making sure everything is downloaded correctly on your non-gpu machine, with git submodule update --init --recursive

you can also check the git commits against main:
[image]

and for data:
[image]

xtli12, 2023-08-07:
The problem has been solved by thoroughly checking those libraries against the main one. Thanks for your kind assistance!

closed 2023-08-07 by xtli12