facebookresearch/home-robot #324: Env "HabitatOpenVocabManipEnv" running speed stay slow.

ConnerQiu, 2023-08-07:
Inside docker built based on the provided challenge DockerFile, I tested the running speed of "HabitatOpenVocabManipEnv"
I initialized a single env with the config ""projects/habitat_ovmm/configs/env/hssd_demo.yaml" and the action is made by the RandomAgent. It turn out be only 32 fps , which is quite slow. While in the white paper, it is report that the environment can run around 300 fps with HABO (page 22). 

Is there any possible source that slow the speed? The test is completed in a Machine with Intel(R) Xeon(R) Silver 4314 CPU and 4090 GPU. 

Thanks

cpaxton, 2023-08-07:
Could you share the output of nvidia-smi while you are running the docker image?

yvsriram, 2023-08-07:
Also, can you share how you are computing the speed? On page 22, we benchmarked the speed through our training script.

ConnerQiu, 2023-08-07:
@cpaxton @yvsriram Sure, for a better comparison, here is the nvidia-smi output and commanline output when I run the training script with default 20 envs. The fps now is 123. while preivous it was just 13 . I think the GPU-Util is quite low.  

`2023-08-07 17:12:45,228 Resizing observation of ovmm_nav_goal_segmentation: from (640, 480) to (160, 120)
2023-08-07 17:12:45,229 Resizing observation of receptacle_segmentation: from (640, 480) to (160, 120)
2023-08-07 17:12:45,229 Resizing observation of robot_head_depth: from (640, 480) to (160, 120)
2023-08-07 17:12:47,666 agent number of parameters: 9472341
2023-08-07 17:13:51,549 update: 400     fps: 123.188
2023-08-07 17:13:51,550 update: 400     env-time: 9232.576s     pth-time: 1545.183s     frames: 1331200`

`+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 530.30.02              Driver Version: 530.30.02    CUDA Version: 12.1     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                  Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf            Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4090         On | 00000000:51:00.0 Off |                  Off |
|  0%   41C    P2               63W / 450W|  10024MiB / 24564MiB |     11%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce RTX 4090         On | 00000000:C3:00.0 Off |                  Off |
|  0%   40C    P8               32W / 450W|      9MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A    729755      G   /usr/lib/xorg/Xorg                          140MiB |
|    0   N/A  N/A    730861      G   /usr/bin/gnome-shell                         14MiB |
|    0   N/A  N/A    730923      G   /usr/libexec/mutter-x11-frames              103MiB |
|    0   N/A  N/A   1030173      C   python                                     3032MiB |
|    0   N/A  N/A   1030247      G   /opt/conda/envs/home-robot/bin/python       282MiB |
|    0   N/A  N/A   1030248      G   /opt/conda/envs/home-robot/bin/python       228MiB |
|    0   N/A  N/A   1030249      G   /opt/conda/envs/home-robot/bin/python       203MiB |
|    0   N/A  N/A   1030250      G   /opt/conda/envs/home-robot/bin/python       254MiB |
|    0   N/A  N/A   1030251      G   /opt/conda/envs/home-robot/bin/python       214MiB |
|    0   N/A  N/A   1030252      G   /opt/conda/envs/home-robot/bin/python       211MiB |
|    0   N/A  N/A   1030253      G   /opt/conda/envs/home-robot/bin/python       253MiB |
|    0   N/A  N/A   1030254      G   /opt/conda/envs/home-robot/bin/python       275MiB |
|    0   N/A  N/A   1030255      G   /opt/conda/envs/home-robot/bin/python       203MiB |
|    0   N/A  N/A   1030256      G   /opt/conda/envs/home-robot/bin/python       282MiB |
|    0   N/A  N/A   1030257      G   /opt/conda/envs/home-robot/bin/python       248MiB |
|    0   N/A  N/A   1030258      G   /opt/conda/envs/home-robot/bin/python       406MiB |
|    0   N/A  N/A   1030259      G   /opt/conda/envs/home-robot/bin/python       248MiB |
|    0   N/A  N/A   1030260      G   /opt/conda/envs/home-robot/bin/python       219MiB |
|    0   N/A  N/A   1030261      G   /opt/conda/envs/home-robot/bin/python       273MiB |
|    0   N/A  N/A   1030262      G   /opt/conda/envs/home-robot/bin/python       227MiB |
|    0   N/A  N/A   1030263      G   /opt/conda/envs/home-robot/bin/python       305MiB |
|    0   N/A  N/A   1030264      G   /opt/conda/envs/home-robot/bin/python       244MiB |
|    0   N/A  N/A   1030265      G   /opt/conda/envs/home-robot/bin/python       256MiB |
|    0   N/A  N/A   1030266      G   /opt/conda/envs/home-robot/bin/python       351MiB |
|    0   N/A  N/A   1030267      G   /opt/conda/envs/home-robot/bin/python       231MiB |
|    0   N/A  N/A   1030268      G   /opt/conda/envs/home-robot/bin/python       218MiB |
|    0   N/A  N/A   1030269      G   /opt/conda/envs/home-robot/bin/python       279MiB |
|    0   N/A  N/A   1030270      G   /opt/conda/envs/home-robot/bin/python       175MiB |
|    0   N/A  N/A   1030271      G   /opt/conda/envs/home-robot/bin/python       341MiB |
|    0   N/A  N/A   1030272      G   /opt/conda/envs/home-robot/bin/python       217MiB |
|    1   N/A  N/A    729755      G   /usr/lib/xorg/Xorg                            4MiB |
+---------------------------------------------------------------------------------------+`

cpaxton, 2023-08-08:
Another thing to note is that our times are parallelized on multiple instances as per prior work using habitat.

cpaxton, 2023-08-08:
Maybe @JHurricane96 / @ykarmesh  can comment here

cpaxton, 2023-08-10:
@ConnerQiu I believe this is expected behavior; we train multi-cpu/multi-gpu and report higher framerates that way, as per the past works in habitat.

With random actions, 32 fps seems about right because you should see a lot of environment resets. This also probably explains low GPU utilization.

Are you having issues with training?

ykarmesh, 2023-08-11:
@ConnerQiu if your GPU Util is low, I would suggest that you use more parallel envs during training. More parallel envs will also help you in reducing the noise in the training.

cpaxton, 2023-08-14:
Closing as this is expected behavior