facebookresearch/home-robot #325: Unable to submit using eval ai cli

yutojubako, 2023-08-07:
## 🐛 Bug

I would like to submit online according to [this markdown](https://github.com/facebookresearch/home-robot/blob/main/docs/challenge.md).
I executed the above command, but I get the following error message and cannot submit.
```
$ evalai push ovmm_baseline_submission:latest --phase minival
Error: Challenge phase with slug minival does not exist
``` 

I looked through the github of evalai cli, but no such error stood in the issue. Also, the results of running `evalai challenges `were as follows.
```
$ evalai challenges
404 Client Error: Not Found for url: https://eval.ai/api/challenges/challenge/all
``` 
If you know how to solve this problem, please let me know.
## Steps to Reproduce

Steps to reproduce the behavior: Run [online submission section](https://github.com/facebookresearch/home-robot/blob/main/docs/challenge.md#online-submission) as is.

## Expected behavior

## Additional context

cpaxton, 2023-08-07:
Could you make sure you followed these instructions: https://eval.ai/web/challenges/challenge-page/2100/submission

cpaxton, 2023-08-07:
The phase names are a bit unintuitive, and may not be appearing on the EvalAI side.

Check out this PR I just created; it includes examples of the phase commands: https://github.com/facebookresearch/home-robot/pull/326

Hopefully this resolves your issue

cpaxton, 2023-08-07:
Specifically you want to run this:
```
# For minival submissions
# With a public submission, anyone can see your score
evalai push <image>:<tag> --phase neurips-ovmm-minival-2023-2100
# You can also make a private submission - no one but you and the organizers can see the results
evalai push <image>:<tag> --phase neurips-ovmm-minival-2023-2100 --private
```

yutojubako, 2023-08-08:
Thank you for your response. I got to submit it.

closed 2023-08-08 by yutojubako