facebookresearch/home-robot #330: OVMM Challenge Docker Container Ubuntu Version

trangml-mit, 2023-08-08:
## 🐛 Bug

Right now the homerobot-ovmm-challenge-2023 Docker container is using Ubuntu 22.04, which is problematic, as it does not support ROS Noetic.

## Expected behavior

The docker container should be Ubuntu 20.04, consistent with the preliminary README instructions [https://github.com/facebookresearch/home-robot#preliminary](https://github.com/facebookresearch/home-robot#preliminary)

> HomeRobot requires Python 3.9. Installation on a workstation requires [conda](https://docs.conda.io/projects/conda/en/latest/user-guide/install/linux.html) and [mamba](https://mamba.readthedocs.io/en/latest/user_guide/mamba.html). Installation on a robot assumes Ubuntu 20.04 and [ROS Noetic](http://wiki.ros.org/noetic).

cpaxton, 2023-08-10:
@trangml-mit do you need Noetic for simulation evaluations?

trangml-mit, 2023-08-10:
Yes, for our solution we plan on using a mapping technique/software that needs Noetic.

cpaxton, 2023-08-11:
Is there any chance of building from source for now, or using ROS2? We'll work on getting that supported

rpartsey, 2023-08-13:
Hi @trangml-mit, thanks for reporting this!
Can you, please, try using `fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023-ubuntu20.04` Docker image as a base image for your submission and let us know if it resolves your issue?

cpaxton, 2023-08-14:
Should be resolved; re-open if there are any issues

trangml-mit, 2023-08-15:
Sorry, just got around to testing this, but it doesn't appear that I can pull the `fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023-ubuntu20.04` image, and I don't see it when I view it here [https://hub.docker.com/r/fairembodied/habitat-challenge/tags](https://hub.docker.com/r/fairembodied/habitat-challenge/tags). Was it made public?

cpaxton, 2023-08-15:
Reopening the issue until we can verify it's fixed

cpaxton, 2023-08-15:
@rpartsey might not have been pushed?

rpartsey, 2023-08-15:
Oh, sorry for that @trangml-mit and @cpaxton. Should be available now.

closed 2023-08-16 by trangml-mit