facebookresearch/home-robot #332: 人噢波特

Chortine, 2023-08-10:
## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

## Expected behavior

## Additional context

yvsriram, 2023-08-10:
Can you please add a visualization and share more details (scene id/episode id)?

yvsriram, 2023-08-10:
The `train` episodes do not come with valid episode inits. You can append `habitat.task.episode_init=False` to your command to randomly sample a valid init.

cpaxton, 2023-08-10:
@yvsriram should we open a PR to make this a default?

cpaxton, 2023-08-14:
@Chortine is the issue resolved?

Chortine, 2023-08-15:
> @Chortine is the issue resolved?

Yes, it now can run. Thanks !

Chortine, 2023-08-15:
> > @Chortine is the issue resolved?
> 
> Yes, it now can run. Thanks !

The command I run is:
`python projects/habitat_ovmm/eval_baselines_agent.py habitat.dataset.episode_ids="[128]" habitat.dataset.split="train" habitat.task.episode_init=False`

However, it keeps producing this warning:

[image]

yvsriram, 2023-08-15:
Yes, I am seeing this on my end as well when running eval with **train** episodes. The spawning seems fine but my guess is that start position from episode is being used as an endpoint for shortest path computation somewhere (mostly outside of OVMM code).

I will check:
1. where this is.
2. if it will affect OVMM evaluations

Meanwhile, feel free to use `HABITAT_SIM_LOG=quiet` to disable warnings

yvsriram, 2023-08-25:
Pushed a fix for this and also removed `episode_init` option (you needn't set/unset it anymore). Let me know if you are still facing issues running evaluations on `train`.

cpaxton, 2023-08-25:
@Chortine is your issue stiill a problem?

cpaxton, 2023-08-29:
Closing as we pushed a fix and there's no further updates