facebookresearch/home-robot #334: OVMM Vocabulary Questions

trangml-mit, 2023-08-10:
For the mission prompts in all the OVMM challenge, will the names of the objects and receptacles always be from the list of classes in the [https://github.com/facebookresearch/home-robot/blob/main/src/home_robot/home_robot/perception/constants.py](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot/home_robot/perception/constants.py) file?

cpaxton, 2023-08-10:
No, they may be anything. We're keeping the specific test set secret, although it's divided between seen and unseen categories, and all objects are "manipulable" - so nothing huge/weird.

trangml-mit, 2023-08-10:
Will there be objects that were completely unseen in the current OVMM dataset, or will it just be different names for the objects that we've already seen?

cpaxton, 2023-08-10:
It will be new objects, completely unseen.
I think it's worth running your method on the test-standard dataset, but validation is also a good example - it similarly contains held-out object categories.

cpaxton, 2023-08-11:
We are using a closed set of *receptacle* categories, which is here: https://github.com/facebookresearch/home-robot/blob/4e3149117a1d237a8d7a35eb2f92dec3145dc0bd/projects/real_world_ovmm/configs/example_cat_map.json#L132

trangml-mit, 2023-08-14:
Thank you, that helps

closed 2023-08-14 by trangml-mit