facebookresearch/home-robot #341: Can't start docker

xtli12, 2023-08-22:
## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:
I followed the [Participation Guidelines](https://github.com/facebookresearch/home-robot/blob/main/docs/challenge.md) to build my docker image, but when I followed the
```
docker system prune -a
./scripts/test_local.sh --docker-name ovmm_baseline_submission
```
I've found that I can't use docker on my machine anymore, the error is:
```
ERROR: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
```
and I can't restart my docker by runing `systemctl start docker`:
```
(base) lxt21@ubuntu:~$ systemctl restart docker
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'docker.service'.
Authenticating as: user
Password: 
==== AUTHENTICATION COMPLETE ===
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
```
then I try to reinstall the docker following [the official guideline](https://docs.docker.com/engine/install/ubuntu/),  but after I ran 
```
sudo apt-get install docker-ce docker-ce-cli containerd.io docker-buildx-plugin docker-compose-plugin
```
the error is 
```
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2023-08-21 14:28:51 UTC; 10ms ago
     Docs: https://docs.docker.com
  Process: 62813 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
 Main PID: 62813 (code=exited, status=1/FAILURE)
dpkg: error processing package docker-ce (--configure):
 installed docker-ce package post-installation script subprocess returned error exit status 1
```
It appears that the failure is due to the inability to restart Docker. Despite my attempts to modify the daemon file and subsequently reload it, I still can't successfully restart Docker. Could you please provide guidance on how to resolve this issue?

cpaxton, 2023-08-23:
Make sure you can run docker without sudo:
```
 docker run hello-world
  # this one might work
 sudo docker run hello-world

```

cpaxton, 2023-08-23:
Also you usually need sudo for
```
sudo systemctl restart docker
```
might want to try restarting your machine. looks like there's something odd in your setup, i'm not sure how to help

cpaxton, 2023-08-23:
You should not modify the daemon file - at this point, you might want to purge docker from your system and reinstall from scratch + make sure you can run the hello-world docker example above

cpaxton, 2023-09-05:
Is this still an issue? Closing due to lack of info/activity

closed 2023-09-05 by cpaxton