facebookresearch/home-robot #347: /hssd-hab/hssd-hab-uncluttered.scene_dataset_config.json| This file does not exist

FUIGUIMURONG, 2023-08-28:
When I run 
`python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml`
There is a error:
`Loading pretrained CLIP
/home/sdp/mambaforge/envs/home-robot/lib/python3.9/site-packages/torch/functional.py:504: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /home/conda/feedstock_root/build_artifacts/pytorch-recipe_1673738523684/work/aten/src/ATen/native/TensorShape.cpp:3190.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2023-08-28 17:18:19,583 Initializing dataset OVMMDataset-v0
2023-08-28 17:18:20,965 initializing sim OVMMSim-v0
[17:18:21:584739]:[Metadata] StageAttributesManager.cpp(86)::registerObjectFinalize : Render asset template handle : data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json specified in stage template with handle : data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json does not correspond to any existing file or primitive render asset.  Aborting.
Traceback (most recent call last):
  File "/home/sdp/home-robot/projects/habitat_ovmm/eval_baselines_agent.py", line 98, in <module>
    metrics = evaluator.evaluate(
  File "/home/sdp/home-robot/projects/habitat_ovmm/evaluator.py", line 458, in evaluate
    self._env = create_ovmm_env_fn(self.config)
  File "/home/sdp/home-robot/projects/habitat_ovmm/utils/env_utils.py", line 34, in create_ovmm_env_fn
    habitat_env = env_class(config=habitat_config, dataset=dataset)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 117, in __init__
    base_env = RLTaskEnv(config=config, dataset=dataset)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 43, in __init__
    super().__init__(config, dataset)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 382, in __init__
    self._env = Env(config, dataset)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 114, in __init__
    self._sim = make_sim(
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 82, in __init__
    super().__init__(config)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 289, in __init__
    super().__init__(self.sim_config)
  File "<attrs generated init habitat_sim.simulator.Simulator>", line 11, in __init__
  File "/home/sdp/mambaforge/envs/home-robot/lib/python3.9/site-packages/habitat_sim-0.2.4-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 118, in __attrs_post_init__
    self.__set_from_config(self.config)
  File "/home/sdp/mambaforge/envs/home-robot/lib/python3.9/site-packages/habitat_sim-0.2.4-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 227, in __set_from_config
    self._config_backend(config)
  File "/home/sdp/mambaforge/envs/home-robot/lib/python3.9/site-packages/habitat_sim-0.2.4-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 200, in _config_backend
    super().__init__(config.sim_cfg, config.metadata_mediator)
AssertionError: ESP_CHECK failed: Missing (at least) one of scene dataset attributes, stage attributes, or dataset scene attributes for scene 'data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json'.  Likely an invalid scene name.`
I appriciate if you give me some advice.

yvsriram, 2023-08-28:
Does this file exist for you: `data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json`? If yes, can you please try: `cd data/hssd-hab` and `git lfs pull`.

Did you use the install script at: `https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/install.sh`? If you are using this script, this should already be done for you.

Also, can you share the `hssd-hab` commit you are using? `cd data/hssd-hab` and `git log`.

FUIGUIMURONG, 2023-08-28:
Thanks for repling.
The file: data/hssd-hab/scenes-uncluttered/104348361_171513414.scene_instance.json does not exist. 
To be specific, I have some questions with git on the remote server, so I git clone the hssd-hab on my local computer and uploader the dataset to the remote server. Because the dataset is too large, I create the link in data/hssd-hab that points to the uploaded dataset.
1.As you advise, I just try: cd data/hssd-hab and git lfs pull. Weirdly, it has no response.
2.This is the latest gitlog:
`(home-robot) sdp@xdy-Super-Server:~/home-robot/data/hssd-hab$ git log
commit b643698a372c9b64af8995d7295d28ba8b7dfe99 (HEAD -> main, origin/main, origin/HEAD)
Author: Chris Paxton <cpaxton@fb.com>
Date:   Wed Aug 16 13:53:06 2023 -0400

    Updates for fixing head tilt + motion in smaller kitchen (#241)

commit 50780f78a4fb43b482b65ce360afcc2e101af998
Author: Xiaohan Zhang <43056077+keke-220@users.noreply.github.com>
Date:   Wed Aug 16 10:18:52 2023 -0400

    Add quality of life checks to make sure robots are homed and initalized right (#338)
    
    * Add quality of life checks for experimenters to make sure robots are
    homed and working properly
    
    * add a note about creating a catkin workspace
    
    ---------
    
    Co-authored-by: Chris Paxton <cpaxton@fb.com>

commit 2972734c7b0038ee84ab272e3e0a4454d1cd874f
Author: Sriram Yenamandra <yvsriram99@gmail.com>
Date:   Tue Aug 15 10:22:30 2023 -0400

    minor fix info is not Nnne (#337)

commit 28149f5255224760e9c03248576cd5f433f1510d
Author: Sriram Yenamandra <yvsriram99@gmail.com>
Date:   Mon Aug 14 16:56:16 2023 -0400

    add README for installing perception modules (#315)
    
    * add README for installing perception modules
    
    * Update install_deps.sh
    
    * Mention GroundedSAM in README.md

commit 647f9975121f507dc48913bb8ba129028847e8e9
Author: Ruslan <partsey2412@gmail.com>
Date:   Mon Aug 14 19:28:40 2023 +0300

    added ubuntu20.04 base challenge Docker image; updated challenge readme (#335)

commit 4e3149117a1d237a8d7a35eb2f92dec3145dc0bd
Author: Chris Paxton <cpaxton@fb.com>
Date:   Wed Aug 9 11:07:19 2023 -0400

    Extra note about python (#331)
    
    * Add extra note about python 3.9
    
    * extra note about python and dependencies

commit 61743670d0900ef8ce6753dafdd1bd348822e553
Author: Chris Paxton <cpaxton@fb.com>
Date:   Mon Aug 7 15:43:22 2023 -0400

    Fix bug with agent evaluation (#327)
    
    * add __init__
    
    * fix agent command because of condition in ovmm perception

commit 90bef70ef8642c601b1e054b6fc5d7da8b6659ff
Author: Chris Paxton <cpaxton@fb.com>
Date:   Mon Aug 7 12:07:00 2023 -0400

    Add specific phase commands since they are misleading (#326)
    
    * add phase example commands
    
    * more detailed commands
    
    * Update the documentation`

yvsriram, 2023-08-28:
This is `git log` for `home-robot`. I wanted to see this for the `hssd-hab` repo. 

Since the file does not exist, most likely you are on the wrong `hssd-hab` commit. You should use the commits linked from `home-robot` [.gitmodules](https://github.com/facebookresearch/home-robot/blob/main/.gitmodules). I would advise downloading the git submodules locally: 

1. Clone `home-robot` locally
2. `git submodule update --init data/hssd-hab` (or run [https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/install.sh](install.sh))

FUIGUIMURONG, 2023-08-28:
Thanks for your advice. I want to figure it more clear.
1.I change to the original `hssd-hab` directory and use` git log`(not the linked directory). And it shows like the following. Does the `git log` for `hssd-hab`?:
```python
(home-robot) sdp@xdy-Super-Server:/data/sdp/data/OVMM/hssd-hab$ git log
commit 66bc2f1f15e5e7fbf5d8db4ad0458b569fac352d (HEAD -> main, tag: v0.2.4, origin/main, origin/HEAD)
Author: Mukul Khanna <mukul18khanna@gmail.com>
Date:   Wed Aug 16 19:50:40 2023 +0530

    Uploads recompressed object GLBs

commit 5f7980f98a7adddb383e0ef4af8293b08eb0a89c
Author: Mukul Khanna <mukul18khanna@gmail.com>
Date:   Wed Aug 16 13:14:03 2023 +0530

    Adds semantic lexicon and csv file with object metadata

commit 53ade654035c9f165c63ba8496ee82a661c57f67
Author: Mukul Khanna <mukul18khanna@gmail.com>
Date:   Wed Aug 16 13:08:43 2023 +0530

    Adds train/val scene splits yaml

commit 4df914754fccb77a307d69b638434327e7200e9c
Author: Mukul Khanna <mukulkhanna@users.noreply.huggingface.co>
Date:   Mon Aug 14 17:17:47 2023 +0000
``` 
2. Is the data structure for hssd-hab?
```python
(home-robot) sdp@xdy-Super-Server:/data/sdp/data/OVMM/hssd-hab$ ll
总用量 76
drwxrwxr-x  7 sdp sdp  4096 8月  27 22:40 ./
drwxrwxr-x  4 sdp sdp  4096 8月  27 17:25 ../
drwxrwxr-x  8 sdp sdp  4096 8月  27 20:04 .git/
-rw-rw-r--  1 sdp sdp  2462 8月  27 15:56 .gitattributes
-rw-rw-r--  1 sdp sdp    11 8月  27 15:56 .gitignore
-rw-rw-r--  1 sdp sdp   664 8月  27 15:56 hssd-hab.scene_dataset_config.json
drwxrwxr-x 21 sdp sdp  4096 8月  27 22:35 objects/
-rw-rw-r--  1 sdp sdp  3199 8月  27 15:56 README.md
drwxrwxr-x  2 sdp sdp 16384 8月  27 22:40 scenes/
-rw-rw-r--  1 sdp sdp  3884 8月  27 15:57 scene_splits.yaml
drwxrwxr-x  2 sdp sdp  4096 8月  27 22:40 semantics/
drwxrwxr-x  2 sdp sdp 20480 8月  27 22:42 stages/
``` 
3. I used the `git clone https://huggingface.co/datasets/hssd/hssd-hab` to download the` hss-hab` . Is the right way to download the dataset?

yvsriram, 2023-08-28:
1. Yes, that is the `git log`. This is the `main` branch which does not have uncluttered scenes we are using.
2. This is actually not the right way as it downloads the `main` branch. Again, I would advise you to use submodules and [download script](https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/install.sh) or commands from it so that you download the right combination of assets. You can run these locally and then upload them if you are having git issues on your server.

FUIGUIMURONG, 2023-08-29:
I do the git `submodule update --init --recursive` , and the` git submodule update --init data/hssd-hab` to download the `hssd-hab`. The similar way to download the dataset `/dataset/ovmm` and `object`. When I run the` python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml`, it shows like this:
1.```python
[18:01:16:329175]:[Sim] Simulator.cpp(932)::recomputeNavMesh : reconstruct navmesh successful
Traceback (most recent call last):
  File "/home/sdp/home-robot/projects/habitat_ovmm/eval_baselines_agent.py", line 97, in <module>
    metrics = evaluator.evaluate(
  File "/home/sdp/home-robot/projects/habitat_ovmm/evaluator.py", line 459, in evaluate
    return self.local_evaluate(agent, num_episodes)
  File "/home/sdp/home-robot/projects/habitat_ovmm/evaluator.py", line 259, in local_evaluate
    observations, done = self._env.reset(), False
  File "/home/sdp/home-robot/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py", line 116, in reset
    habitat_obs = self.habitat_env.reset()
  File "/home/sdp/mambaforge/envs/home-robot/lib/python3.9/site-packages/gym/core.py", line 283, in reset
    return self.env.reset(**kwargs)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/utils/gym_adapter.py", line 300, in reset
    obs = self._env.reset()
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 54, in reset
    observations = super().reset()
  File "/home/sdp/mambaforge/envs/home-robot/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 422, in reset
    return self._env.reset()
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 262, in reset
    observations = self.task.reset(episode=self.current_episode)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/sub_tasks/nav_to_obj_task.py", line 53, in reset
    obs = super().reset(episode)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/sub_tasks/nav_to_obj_task.py", line 267, in reset
    return self._get_observations(episode)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_task.py", line 193, in _get_observations
    task_obs = self.sensor_suite.get_observations(
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/simulator.py", line 229, in get_observations
    return Observations(self.sensors, *args, **kwargs)
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/simulator.py", line 123, in __init__
    data = [
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/simulator.py", line 124, in <listcomp>
    (uuid, sensor.get_observation(*args, **kwargs))
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/sub_tasks/nav_to_obj_sensors.py", line 118, in get_observation
    obs[..., 1] = self._get_obs_channel(
  File "/home/sdp/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/sub_tasks/nav_to_obj_sensors.py", line 87, in _get_obs_channel
    handle = self._sim.receptacles[
KeyError: 'receptacle_mesh_1d5a78b46d32bf41584c800a0dfa2536d7f0b395.0000'
``` 
2. I check the `git log`  for the three dataset:
1)As for `ovmm`:
```python
(base) sdp@xdy-Super-Server:~/home-robot/data/datasets/ovmm$ git log
commit 4ee7e6b36c1a0b01b08354acdf909204436f9d14 (HEAD)
Author: Arun Ramachandran <arun.jhurricane@gmail.com>
Date:   Sun Jun 18 21:56:45 2023 -0400

    Release episodes

commit 3fc838aa95d14da91ce69fd5243bf38e399ba2a9
Author: Arun Ramachandran <arun.jhurricane@gmail.com>
Date:   Mon Jun 19 01:52:36 2023 +0000

    initial commit
``` 
2) As for` hssd-hab`:
```python
(base) sdp@xdy-Super-Server:~/home-robot/data/hssd-hab$ git log
commit 22a8c888524f37d3e11dd8f98a21bcba8cad4e12 (HEAD)
Author: Manolis Savva <manolis.savva@gmail.com>
Date:   Tue Jun 20 08:45:59 2023 -0700

    add receptacle plys

commit d69278fe1247c27411f2605c38bbf0391f34c870
Author: Manolis Savva <manolis.savva@gmail.com>
Date:   Mon Jun 19 16:22:48 2023 -0700

    add colliders and receptacles

commit 7f2834a47e6798593488cfea95b41049e0288b4f
Author: Manolis Savva <manolis.savva@gmail.com>
Date:   Mon Jun 19 13:54:32 2023 -0700

    add uncluttered scenes, navmeshes

commit d8de88bfc22e89ce911f82df994838e5d1815b2e (tag: v0.2.3)
Author: Manolis Savva <manolis.savva@gmail.com>
Date:   Wed Jun 14 20:16:33 2023 -0700

    Update README.md
``` 
3) As for the `object`:
```python
(base) sdp@xdy-Super-Server:~/home-robot/data/objects$ git log
commit d11b3a0edd401f285d04d9c7bca493f45e0297b8 (HEAD)
Author: Arun Ramachandran <JHurricane96@users.noreply.huggingface.co>
Date:   Wed Jun 21 14:19:04 2023 +0000

    missing_objects (#1)
    
    
    - add missing object files (6201f4b61a85a794ecc79acfa1359781cce26114)
    
    
    Co-authored-by: sriram <yvsriram@users.noreply.huggingface.co>

commit 1863868d39af7d92f4af46fd899e8bd689c177a4
``` 
I cannot locate which dataset miss something that result  ` KeyError: 'receptacle_mesh_1d5a78b46d32bf41584c800a0dfa2536d7f0b395.0000'`. I really hope you can give me some advice.

yvsriram, 2023-08-29:
Yes, now it looks like the dataset you downloaded is incompatible with the `habitat-lab` version. The home-robot on your local machine (where you downloaded the data) uses an older home-robot commit. Can you please run `git pull && git submodule update --init --recursive` locally to download the latest datasets?

FUIGUIMURONG, 2023-08-29:
I run `git pull && git submodule update --init --recursive` as you suggest , and it shows like this:
```python
(base) sdp@xdy-Super-Server:~/home-robot$ git pull && git submodule update --init --recursive
remote: Enumerating objects: 426, done.
remote: Counting objects: 100% (426/426), done.
remote: Compressing objects: 100% (202/202), done.
remote: Total 426 (delta 252), reused 371 (delta 218), pack-reused 0
Received objects: 100% (426/426), 6.11 MiB | 27.00 KiB/s, done.
Processing deltas: 100% (252/252), completed with 71 local objects.
From https://github.com/facebookresearch/home-robot
   b643698a..f4134052  main                 -> origin/main
   727230fc..04c90b9d  alexsax/scannet      -> origin/alexsax/scannet
 * [New branch]            alexsax/scannet_data -> origin/alexsax/scannet_data
 * [New branch]            cpaxton/voxel-rrt    -> origin/cpaxton/voxel-rrt
   4baec203..d210deec  goat                 -> origin/goat
 * [New branch]            mchang/abnb1         -> origin/mchang/abnb1
 * [New branch]            new_ovmm_episodes    -> origin/new_ovmm_episodes
   4b021514..2d8a4062  ray_tracing          -> origin/ray_tracing
 * [New lable]            v0.1.1               -> v0.1.1
Fetching submodule src/third_party/habitat-lab
来自 https://github.com/facebookresearch/habitat-lab
   12e735d2..9ccca099  ovmm              -> origin/ovmm
   1be042fc..9efa20d1  SIRo              -> origin/SIRo
   b2011c70..cafdd508  SIRo_basevelhuman -> origin/SIRo_basevelhuman
   b4a8bc9a..67a07cef  SIRo_dev_add_nn_policy_backup_backup_integrate_hrl_with_ll -> origin/SIRo_dev_add_nn_policy_backup_backup_integrate_hrl_with_ll
   a60f8a39..1f8224e9  SIRo_dev_add_nn_policy_backup_backup_integrate_hrl_with_ll_integrate_nn_nav -> origin/SIRo_dev_add_nn_policy_backup_backup_integrate_hrl_with_ll_integrate_nn_nav
 * [New branch]            dataset-pickle    -> origin/dataset-pickle
 * [New branch]            hab3_cleanup      -> origin/hab3_cleanup
   eb39df34..462c37b6  main              -> origin/main
   20848ed1..b8694e35  ovmm_main_merge   -> origin/ovmm_main_merge
   98265ddb..bfc784e5  social_nav_randwalk_integration -> origin/social_nav_randwalk_integration
 * [New branch]            social_nav_randwalk_integration_oracle -> origin/social_nav_randwalk_integration_oracle
Updating b643698a..f4134052
Fast-forward
 .gitignore                                                                             |   19 +-
 data/datasets/ovmm                                                                     |    2 +-
 data/hssd-hab                                                                          |    2 +-
 projects/habitat_ovmm/configs/agent/heuristic_agent.yaml                               |    4 +
 projects/habitat_ovmm/configs/agent/rl_agent.yaml                                      |    4 +
 projects/offline_mapping/build_map.py                                                  |    4 +-
 projects/pick_location_real_robot.py                                                   |  175 ----------------
 projects/real_world_ovmm/README.md                                                     |    6 +-
 projects/real_world_ovmm/build_3d_map.py                                               |  320 +++++++++++++++++++++++-----
 projects/real_world_ovmm/eval_episode.py                                               |    3 +-
 projects/real_world_ovmm/experimental/demo_fastsam.py                                  |   50 +++++
 projects/real_world_ovmm/experimental/demo_sam.py                                      |   71 +++++++
 projects/real_world_ovmm/experimental/demo_zerodepth.py                                |   54 +++++
 projects/real_world_ovmm/experimental/execute_trajectory.py                            |  125 +++++++++++
 projects/real_world_ovmm/experimental/goals.txt                                        |    3 +
 projects/real_world_ovmm/experimental/mapping.py                                       |  188 +++++++++++++++++
 projects/real_world_ovmm/experimental/trajectories.txt                                 |    3 +
 projects/real_world_ovmm/pick_cup_real_robot.py                                        |   28 ---
 src/home_robot/environment.yml                                                         |    1 +
 src/home_robot/home_robot/agent/objectnav_agent/objectnav_agent.py                     |    3 +
 src/home_robot/home_robot/agent/objectnav_agent/objectnav_agent_module.py              |    3 +
 src/home_robot/home_robot/agent/ovmm_agent/__init__.py                                 |    5 +
 src/home_robot/home_robot/agent/ovmm_agent/ovmm_perception.py                          |   14 +-
 src/home_robot/home_robot/agent/ovmm_agent/ppo_agent.py                                |   18 +-
 src/home_robot/home_robot/core/interfaces.py                                           |   10 +
 src/home_robot/home_robot/datasets/__init__.py                                         |    0
 src/home_robot/home_robot/datasets/scannet/.gitignore                                  |   11 +
 src/home_robot/home_robot/datasets/scannet/__init__.py                                 |    6 +
 src/home_robot/home_robot/datasets/scannet/data/batch_load_scannet_data.py             |  197 ++++++++++++++++++
 src/home_robot/home_robot/datasets/scannet/data/extract_instance_images.py             |  155 ++++++++++++++
 src/home_robot/home_robot/datasets/scannet/data/extract_posed_images.py                |  240 +++++++++++++++++++++
[... cut ...]
Terminating
error: Your local modifications to the following files will be overwritten by the checkout operation：
        habitat-lab/habitat/robots/fetch_robot.py
        habitat-lab/habitat/robots/mobile_manipulator.py
Please commit or stash your changes before switching branches.
Terminating
Unable to check out '2c435284686c1f598ac311c33211c8af9b6e1017' in submodule path 'data/hssd-hab'.
Unable to check out '9ccca099742df667c4e4fc43e204abd8c60e6028' in submodule path 'src/third_party/habitat-lab'.
```

FUIGUIMURONG, 2023-08-29:
I can operate on the remote server, sorry about not make it clear with you.
I do the 3. operation as you instruct. It shows the same like this:
```python
(base) sdp@xdy-Super-Server:~/home-robot$ cd src/third_party/habitat-lab
(base) sdp@xdy-Super-Server:~/home-robot/src/third_party/habitat-lab$ git reset --hard HEAD
HEAD 现在位于 12e735d2 use unique recep names, rename navmesh folder (#1541)
(base) sdp@xdy-Super-Server:~/home-robot/src/third_party/habitat-lab$ git submodule update --init --recursive
(base) sdp@xdy-Super-Server:~/home-robot/src/third_party/habitat-lab$ cd ../..
(base) sdp@xdy-Super-Server:~/home-robot/src$ cd ..
(base) sdp@xdy-Super-Server:~/home-robot$ git submodule update --init --recursive
error: 工作区中下列未跟踪的文件将会因为检出操作而被覆盖：
        navmeshes/scenes-uncluttered/stretch_robot/104348361_171513414.navmesh
        navmeshes/scenes-uncluttered/stretch_robot/104348361_171513414.pkl
请在切换分支前移动或删除。
正在终止
子模组路径 'src/third_party/habitat-lab'：检出 '9ccca099742df667c4e4fc43e204abd8c60e6028'
无法在子模组路径 'data/hssd-hab' 中检出 '2c435284686c1f598ac311c33211c8af9b6e1017'
```

yvsriram, 2023-08-29:
Ok great.
In that case, you can just delete the two files:
`navmeshes/scenes-uncluttered/stretch_robot/104348361_171513414.navmesh` and `navmeshes/scenes-uncluttered/stretch_robot/104348361_171513414.pkl`. These must have been generated when you started evaluations. 

After delete these, you can run `git submodule update --init --recursive`.

FUIGUIMURONG, 2023-08-29:
I do delete the two files and conduct `git submodule update --init --recursive`. But the weird thing happens, it shows the deleted files still exist. I have checked these two files have been deleted.
```python
(base) sdp@xdy-Super-Server:~/home-robot$ git submodule update --init --recursive
error: The following untracked files in the working directory will be overwritten by the checkout operation：
        navmeshes/scenes-uncluttered/stretch_robot/104348361_171513414.navmesh
        navmeshes/scenes-uncluttered/stretch_robot/104348361_171513414.pkl
Please move or delete before switching branches.
Terminating
Unable to check out '2c435284686c1f598ac311c33211c8af9b6e1017' in submodule path 'data/hssd-hab'.
(base) sdp@xdy-Super-Server:~/home-robot$ cd data/hssd-hab/navmeshes/scenes-uncluttered/stretch_robot/
(base) sdp@xdy-Super-Server:~/home-robot/data/hssd-hab/navmeshes/scenes-uncluttered/stretch_robot$ ls
(base) sdp@xdy-Super-Server:~/home-robot/data/hssd-hab/navmeshes/scenes-uncluttered/stretch_robot$ 
```

yvsriram, 2023-08-29:
These would be inside `data/hssd-hab`. Please add `data/hssd-hab/` to start of these paths

FUIGUIMURONG, 2023-08-29:
> ```python
> git submodule update --init --recursive
> ```

Sorry I donnot catch your meaning, The data/hssd-hab/ is actually the start like [https://github.com/facebookresearch/home-robot/issues/347#issuecomment-1697377636](url) shows .  Would you please give me more specific instructions.

yvsriram, 2023-08-29:
Yes, it looks like you have deleted these files inside `data/hssd-hab`. Not sure why git says that these are present. Can you share the output of `cd data/hssd-hab && git status`.

FUIGUIMURONG, 2023-08-29:
> 

I have tried, but the same problem happens. I decide to delete the `hssd-hab` and re-download it to see if it's ok

cpaxton, 2023-08-30:
Will close this and continue discussion in the other issue

closed 2023-08-30 by cpaxton