facebookresearch/home-robot #353: The code cannot from numba.utils import IS_PY3

FUIGUIMURONG, 2023-08-30:
## 🐛 Bug

The code cannot from numba.utils import IS_PY3 

## Steps to Reproduce

Steps to reproduce the behavior:
When I run the demo with `python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml`
 It reports:
```python
Could not import from numba, which means that some
parts of this code may run MUCH more slowly.  You
may wish to install numba
``` 
My envirnment installed the `llvmlite-0.40.1 numba-0.57.1`. I watch the warning origin, it happens because the code runs to `from numba.utils import IS_PY3`. I check it that there is no IS_PY3 in new version Numba anymore. 

And when I do the  Evaluation on complete episode dataset with GT semantics by running `python projects/habitat_ovmm/eval_baselines_agent.py`. The log indicates that it will take 72 hours for complete evaluation. I am using a RTX3090. It seems too slow.

cpaxton, 2023-08-30:
What did you use to create the environment? We just merged a PR which hopefully makes the environment.yml files a bit more reliable - we had some problems before

cpaxton, 2023-08-30:
Or actually we did not merge it yet; PR is here https://github.com/facebookresearch/home-robot/pull/340

You could check out this branch + create env via `mamba`, run ./install_deps.sh, and see if it works?

closed 2023-08-31 by FUIGUIMURONG