facebookresearch/home-robot #357: About evaluation

FUIGUIMURONG, 2023-08-31:
## 🚀 Feature

Would you please describe the evaluation more clearly?
## Motivation

When I do the evaluation using `projects/habitat_ovmm/eval_baselines_agent.py`, I do not know the exact meaning of the `argument`  to be added such as `"--evaluation_type"`  and` "--agent_type"`. Can you give more detailed explanation or is there some docs to recommend?

yvsriram, 2023-08-31:
Thanks for the issue!

Yes, it looks like we missed adding descriptions for these command line arguments. I will add them now.

`evaluation_type` is used to toggle between `local` and `remote`. `remote` is only used by us for running evaluations on EvalAI. So you should always keep this as `local` (the default value).

`agent_type` is used to toggle between our baseline agents (`--agent_type baseline`) and an agent taking random actions (`--agent_type random`)

FUIGUIMURONG, 2023-08-31:
Thanks for replying. I want to ask other questions.
1. When I run ` python projects/habitat_ovmm/eval_baselines_agent.py habitat.dataset.episode_ids="[151,182]"` following the instruction. It returns `
Metrics:
 {'episode_count': 2.0, 'does_want_terminate': 0.0, 'num_steps': 1250.0, 'find_object_phase_success': 0.0                                                                                                     , 'pick_object_phase_success': 0.0, 'find_recep_phase_success': 0.0, 'overall_success': 0.0, 'partial_suc                                                                                                     cess': 0.0}
`
Is it normal?
2. I also run the```
 # Evaluation on complete episode dataset with GT semantics
python projects/habitat_ovmm/eval_baselines_agent.py
```
It indicates there needs 1199 loops to finish which may take too long time.
Would you give some explanations about it?

yvsriram, 2023-08-31:
1199 is the number of validation episodes - each episode is a different instantiation of the OVMM task - scenes, object/receptacle goals and agent start positions are varied across the episodes.

FUIGUIMURONG, 2023-09-01:
> 1199 is the number of validation episodes - each episode is a different instantiation of the OVMM task - scenes, object/receptacle goals and agent start positions are varied across the episodes.

Thanks, I get it. Is the 1199 episodes a must-follow rule? It takes too long time maybe on or two days which I cannot afford.
And what's your opinion about the first question in https://github.com/facebookresearch/home-robot/issues/357#issuecomment-1700831007 .

yvsriram, 2023-09-04:
For the first question in https://github.com/facebookresearch/home-robot/issues/357#issuecomment-1700831007, yes I get the same numbers. 

You can run the evaluation on a subset of episodes using: `python projects/habitat_ovmm/eval_baselines_agent.py habitat.dataset.episode_indices_range="[0,50]"`. This is especially useful at the time of development.

closed 2023-09-04 by FUIGUIMURONG