facebookresearch/home-robot #361: Submission trouble: How to clone hssd-hab dataset from huggingface in dockerfile

YuxinCHENGPE, 2023-09-02:
## 🚀 Feature
Having problem while buiding images for submission caused by hssd-hab clone from huggingface, detailed as follows:
[image]
[image]

Thanks.

yvsriram, 2023-09-08:
Hi, cloning the `hssd-hab` dataset failed for you. Is this issue specific to the docker evaluation or is the normal evaluation also failing for you ([demo command](https://github.com/facebookresearch/home-robot/tree/main/projects/habitat_ovmm#demo-setup))?

yvsriram, 2023-09-08:
Also, may I know how you downloaded the datasets (which script did you run for the outputs in the second image)?

YuxinCHENGPE, 2023-09-09:
> Finally, can you check if the file exists for you - is the data not downloaded at all or is the docker having trouble accessing it?

Hi, thanks for your reply. This issue also happened for the local environment setup on the host machine when I started this challenge. I just followed the dataset setup script to try to download it on the docker (add the command in dockerfile). [HOME_ROBOT_ROOT/projects/habitat_ovmm/install.sh](https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md). I think the clone request of this dataset may needs the huggingface authorization, but I have no idea how to enable it in dockerfile. Can you help me?

yvsriram, 2023-09-09:
Yes, definitely!

Just to clarify - you are saying that you are running into the same error when you run the demo script locally (outside of docker). If yes, your hssd-hab scenes data did not download properly. What is the output you see when you run `git submodule update --init data/hssd-hab` after following the other instructions [here](https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md#scene-dataset-setup).

For docker - you do not need to (and shouldn't) download data inside of your baseline docker. 

Please follow the following steps:
1. Download data inside your `data/` folder using `https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md`
2. Ensure that the demo command runs without errors (this will tell you whether your data/ folder is correctly setup)
3. For testing the docker image using our baseline you can just run steps 2, 5 and 7 here: `https://github.com/facebookresearch/home-robot/blob/main/docs/challenge.md`.

Please do not download data inside the docker. This line [here](https://github.com/facebookresearch/home-robot/blob/912eb36abd91d4467ff4292b6a339ae95714a3a5/projects/habitat_ovmm/scripts/test_local.sh#L23) will make the `data/` folder you setup above automatically accessible inside the docker.

Let us know if you are still running into issues.

YuxinCHENGPE, 2023-09-09:
> Yes, definitely!
> 
> Just to clarify - you are saying that you are running into the same error when you run the demo script locally (outside of docker)? If yes, your hssd-hab scenes data did not download properly. What is the output you see when you run `git submodule update --init data/hssd-hab` after following the other instructions [here](https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md#scene-dataset-setup).
> 
> For docker - you do not need to (and shouldn't) download data inside of your baseline docker.
> 
> Please follow the following steps:
> 
> 1. Download data inside your `data/` folder using https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md
> 2. Ensure that the demo command runs without errors (this will tell you whether your data/ folder is correctly setup)
> 3. For testing the docker image using our baseline you can just run steps 2, 5 and 7 here: https://github.com/facebookresearch/home-robot/blob/main/docs/challenge.md.
> 
> Please do not download data inside the docker. This line [here](https://github.com/facebookresearch/home-robot/blob/912eb36abd91d4467ff4292b6a339ae95714a3a5/projects/habitat_ovmm/scripts/test_local.sh#L23) will make the `data/` folder automatically accessible inside the docker.
> 
> Let us know if you are still running into issues.

Thanks for your reply and I understand. I awared that the local test of docker container will automatically bind the data folder in my host machine into the docker container. But **when I submit the docker image to remote test on the EvalAI**, there will be an exception shown in the first figure, which may be caused by data missing.

yvsriram, 2023-09-09:
Hi, our evaluation docker on EvalAI handles this by creating a symbolic link to a copy of the data. So, if your local evaluation works fine, it isn't expected to cause issues on EvalAI. 

If your EvalAI submission is having issues, we can investigate what is going wrong. If that is the case, can you please share your EvalAI team name with us?

cpaxton, 2023-09-11:
@YuxinCHENGPE just checking, you are still seeing this issue on EvalAI?

YuxinCHENGPE, 2023-09-12:
My teammates will check it today and I believe your explanations are quite clear for us to solve all our problems. Thanks!

closed 2023-09-12 by YuxinCHENGPE