facebookresearch/home-robot #362: Cant run Eval inside docker: Robot urdf location error and keyerror

ruddradev, 2023-09-02:
## 🐛 Bug

I am not able to run the docker container after I clone the home-robot OVMM repository and then download the data. When I run the script to do local evaluations, the docker algorithm is not able to find hab_stretch.urdf under data/robots/ because the folder isn't there. I copied the folder from assets and renamed it to robots but getting a keyerror. 

## Steps to Reproduce

Steps to reproduce the behavior:

1) Clone the home-robot repository.
2) If the data is not downloaded properly, 
```
cd data
git lfs pull
```
3) from projects/habitat_ovmm, run sh ./scripts/test_local.sh --docker-name ovmm_baseline_submission

## Expected behavior

The evaluation should happen.

But instead I am getting the below error
```

AssertionError: ESP_CHECK failed: failed to parse/load URDF file data/robots/hab_stretch/urdf/hab_stretch.urdf
  0%|          | 0/1199 [00:00<?, ?it/s]
```
When I copied the folder from assests into data/robots/

I am getting the below error

```

==========
== CUDA ==
==========

CUDA Version 11.7.1

Container image Copyright (c) 2016-2023, NVIDIA CORPORATION & AFFILIATES. All rights reserved.

This container image and its contents are governed by the NVIDIA Deep Learning Container License.
By pulling and using the container, you accept the terms and conditions of this license:
https://developer.nvidia.com/ngc/nvidia-deep-learning-container-license

A copy of this license is made available in this container at /NGC-DL-CONTAINER-LICENSE for your convenience.

pybullet build time: May 20 2022 19:45:31
Neither `ifconfig` (`ifconfig -a`) nor `ip` (`ip address show`) commands are available, listing network interfaces is likely to fail
/opt/conda/envs/home-robot/lib/python3.9/site-packages/torch/functional.py:504: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /opt/conda/conda-bld/pytorch_1666642975993/work/aten/src/ATen/native/TensorShape.cpp:3190.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2023-09-02 08:38:36,772 Initializing dataset OVMMDataset-v0
2023-09-02 08:38:37,646 initializing sim OVMMSim-v0
[08:38:39:765228]:[Core] ManagedContainerBase.h(329)::checkExistsWithMessage : <Lighting Layout>::getObjectByHandle : Unknown Lighting Layout managed object handle :  . Aborting
2023-09-02 08:38:40,862 Initializing task OVMMNavToObjTask-v0
/opt/conda/envs/home-robot/lib/python3.9/site-packages/gym/spaces/box.py:84: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
  0%|          | 0/1199 [00:00<?, ?it/s]Loading pretrained CLIP
Traceback (most recent call last):
  File "/home-robot/projects/habitat_ovmm/agent.py", line 97, in <module>
    metrics = evaluator.evaluate(
  File "/home-robot/projects/habitat_ovmm/evaluator.py", line 459, in evaluate
    return self.local_evaluate(agent, num_episodes)
  File "/home-robot/projects/habitat_ovmm/evaluator.py", line 259, in local_evaluate
    observations, done = self._env.reset(), False
  File "/home-robot/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py", line 116, in reset
    habitat_obs = self.habitat_env.reset()
  File "/opt/conda/envs/home-robot/lib/python3.9/site-packages/gym/core.py", line 283, in reset
    return self.env.reset(**kwargs)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/utils/gym_adapter.py", line 300, in reset
    obs = self._env.reset()
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 54, in reset
    observations = super().reset()
  File "/opt/conda/envs/home-robot/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 422, in reset
    return self._env.reset()
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 260, in reset
    self.reconfigure(self._config)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 342, in reconfigure
    self._sim.reconfigure(self._config.simulator, self.current_episode)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 260, in reconfigure
    self.robots_mgr.reconfigure(new_scene)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/robot_manager.py", line 102, in reconfigure
    robot_data.robot.reconfigure()
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/robots/mobile_manipulator.py", line 143, in reconfigure
    Manipulator.reconfigure(self)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/robots/manipulator.py", line 111, in reconfigure
    self.sim_obj.update_joint_motor(self.joint_motors[i][0], jms)
KeyError: 33
  0%|          | 0/1199 [00:01<?, ?it/s]
```

## Additional context

FUIGUIMURONG, 2023-09-05:
I have met the same problem like "KeyError: '31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c_:0000|receptacle_mesh_31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c.0000'
  0%|          | 0/1199 [00:03<?, ?it/s]"

cpaxton, 2023-09-05:
Please try training/testing without Docker for now while we work on this

cpaxton, 2023-09-05:
@ruddradev if you are not finding the URDF file, can you make sure that assets/hab_stretch has been properly cloned? it should be created when you run the install_deps.sh script, the main install file in home-robot

ruddradev, 2023-09-06:
@cpaxton : Yes you are right, when I ran the setup to run locally, I got the assets for robot urdf, but am now facing the error I have referred to in Additional context, and also what @FUIGUIMURONG is facing. So the other items of this issue seem to be solved. However I think we should include the steps to download assets and copy them into the docker in the docker setup walkthrough to help others.

yvsriram, 2023-09-09:
Hi @ruddradev 

- For the `KeyError: '31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c_:0000|receptacle_mesh_31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c.0000`, we just updated the [homerobot-ovmm-challenge-2023](https://hub.docker.com/layers/fairembodied/habitat-challenge/homerobot-ovmm-challenge-2023/images/sha256-d5c0f19ad8ae6ab2f3e4e4f1be0905c0db4d0eb693755ace231b48f153b5f215?context=explore) on docker hub. Please try it and let us know if you run into any issues.

- For the error with urdf - we expect the stretch urdf to be downloaded inside the `data` directory by the `install.sh` script [here](https://github.com/facebookresearch/home-robot/blob/912eb36abd91d4467ff4292b6a339ae95714a3a5/projects/habitat_ovmm/install.sh#L39). This [line](https://github.com/facebookresearch/home-robot/blob/912eb36abd91d4467ff4292b6a339ae95714a3a5/projects/habitat_ovmm/scripts/test_local.sh#L23) should mount the `data` directory inside docker making all its contents available.

yvsriram, 2023-09-09:
For using the updated base docker, please run `docker pull fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023`.

kellyiss, 2023-09-09:
I also met this issue "KeyError: '31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c_:0000|receptacle_mesh_31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c.0000' ", even if I don't use the docker image provided. What causes this error?

yvsriram, 2023-09-09:
Hi, this can happen if your `habitat-lab` version is out of date. Can you please check if your `habitat-lab` commit matches the commit tagged here: https://github.com/facebookresearch/home-robot/tree/main/src/third_party? You can use `git submodule update --init src/third_party/habitat-lab` to update habitat-lab code.

kellyiss, 2023-09-09:
Yes, my current habitat-lab commit is 9ccca099742df667c4e4fc43e204abd8c60e6028 which causes this error. And I find that the earlier 8431b5f1f519dee8b1c6f52a2e0e253fa5e1daae commit doesn't have this error.

yvsriram, 2023-09-09:
Can you paste your full stacktrace? Also, please ensure that the data versions match `data/datasets/ovmm` ([ovmm @ f95ca40](https://github.com/facebookresearch/home-robot/blob/main/data/datasets/ovmm)) and `data/hssd-hab` ([hssd-hab @ 2c43528](https://github.com/facebookresearch/home-robot/blob/main/data/hssd-hab)).

kellyiss, 2023-09-10:
After updating data/datasets/ovmm to the newest version, the error disappeared.

closed 2023-09-11 by cpaxton