facebookresearch/home-robot #364: Evaluate docker container locally Problem

FUIGUIMURONG, 2023-09-05:
I want to evaluate your docker container locally using `./scripts/test_local.sh --docker-name ovmm_baseline_submission`. It reports this error. Can you give me some advice?
```python
(base) sdp@xdy-Super-Server:~/home-robot/projects/habitat_ovmm$ ./scripts/test_local.sh --docker-name ovmm_baseline_submission

==========
== CUDA ==
==========

CUDA Version 11.7.1

Container image Copyright (c) 2016-2023, NVIDIA CORPORATION & AFFILIATES. All rights reserved.

This container image and its contents are governed by the NVIDIA Deep Learning Container License.
By pulling and using the container, you accept the terms and conditions of this license:
https://developer.nvidia.com/ngc/nvidia-deep-learning-container-license

A copy of this license is made available in this container at /NGC-DL-CONTAINER-LICENSE for your convenience.

pybullet build time: May 20 2022 19:45:31
Neither `ifconfig` (`ifconfig -a`) nor `ip` (`ip address show`) commands are available, listing network interfaces is likely to fail
Traceback (most recent call last):
  File "/opt/conda/envs/home-robot/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "/opt/conda/envs/home-robot/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/opt/conda/envs/home-robot/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/opt/conda/envs/home-robot/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/opt/conda/envs/home-robot/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
  File "/opt/conda/envs/home-robot/lib/python3.9/http/client.py", line 980, in send
    self.connect()
  File "/opt/conda/envs/home-robot/lib/python3.9/http/client.py", line 1447, in connect
    super().connect()
  File "/opt/conda/envs/home-robot/lib/python3.9/http/client.py", line 946, in connect
    self.sock = self._create_connection(
  File "/opt/conda/envs/home-robot/lib/python3.9/socket.py", line 844, in create_connection
    raise err
  File "/opt/conda/envs/home-robot/lib/python3.9/socket.py", line 832, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home-robot/projects/habitat_ovmm/agent.py", line 91, in <module>
    agent = OpenVocabManipAgent(agent_config)
  File "/home-robot/src/home_robot/home_robot/agent/ovmm_agent/ovmm_agent.py", line 75, in __init__
    self.semantic_sensor = OvmmPerception(config, device_id, self.verbose)
  File "/home-robot/src/home_robot/home_robot/agent/ovmm_agent/ovmm_perception.py", line 77, in __init__
    self._segmentation = DeticPerception(
  File "/home-robot/src/home_robot/home_robot/perception/detection/detic/detic_perception.py", line 128, in __init__
    classifier = get_clip_embeddings(self.metadata.thing_classes)
  File "/home-robot/src/home_robot/home_robot/perception/detection/detic/detic_perception.py", line 60, in get_clip_embeddings
    text_encoder = build_text_encoder(pretrain=True)
  File "/home-robot/src/home_robot/home_robot/perception/detection/detic/Detic/detic/modeling/text/text_encoder.py", line 178, in build_text_encoder
    pretrained_model, _ = clip.load("ViT-B/32", device='cpu')
  File "/opt/conda/envs/home-robot/lib/python3.9/site-packages/clip/clip.py", line 120, in load
    model_path = _download(_MODELS[name], download_root or os.path.expanduser("~/.cache/clip"))
  File "/opt/conda/envs/home-robot/lib/python3.9/site-packages/clip/clip.py", line 59, in _download
    with urllib.request.urlopen(url) as source, open(download_target, "wb") as output:
  File "/opt/conda/envs/home-robot/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/opt/conda/envs/home-robot/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
  File "/opt/conda/envs/home-robot/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/opt/conda/envs/home-robot/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/opt/conda/envs/home-robot/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "/opt/conda/envs/home-robot/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno 111] Connection refused>
```

cpaxton, 2023-09-08:
Hi @FUIGUIMURONG , this is an issue with our docker images, we'll have a fix soon
@yvsriram / @ykarmesh we need to finalize this + push our new docker images https://github.com/facebookresearch/home-robot/pull/370

yvsriram, 2023-09-09:
Hi, we just updated the [homerobot-ovmm-challenge-2023](https://hub.docker.com/layers/fairembodied/habitat-challenge/homerobot-ovmm-challenge-2023/images/sha256-d5c0f19ad8ae6ab2f3e4e4f1be0905c0db4d0eb693755ace231b48f153b5f215?context=explore) on docker hub. Please try it and let us know if you run into any issues.

yvsriram, 2023-09-09:
For using the updated base docker, please run `docker pull fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023`.

FUIGUIMURONG, 2023-09-10:
> For using the updated base docker, please run `docker pull fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023`.

If I use the ubuntu20.04,  should I use `docker pull fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023-ubuntu20.04`

yvsriram, 2023-09-10:
Hi,

You can stick to using the base docker `fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023` which was built for 22.04. You can switch to 20.04 if you are adding some dependency that does not work with 22.04 (eg. https://github.com/facebookresearch/home-robot/issues/330).

FUIGUIMURONG, 2023-09-11:
I evaluate the updated docker container `fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023-ubuntu20.04` locally using `./scripts/test_local.sh --docker-name ovmm_baseline_submission`. It shows the loop time is 1199, not like the instruction in https://github.com/facebookresearch/home-robot/blob/main/docs/challenge.md step 7 shows 10 loop times. 

```python
(base) sdp@xdy-Super-Server:~/home-robot/projects/habitat_ovmm$ ./scripts/test_local.sh --docker-name ovmm_baseline_submission

==========
== CUDA ==
==========

CUDA Version 11.7.1

Container image Copyright (c) 2016-2023, NVIDIA CORPORATION & AFFILIATES. All rights reserved.

This container image and its contents are governed by the NVIDIA Deep Learning Container License.
By pulling and using the container, you accept the terms and conditions of this license:
https://developer.nvidia.com/ngc/nvidia-deep-learning-container-license

A copy of this license is made available in this container at /NGC-DL-CONTAINER-LICENSE for your convenience.

pybullet build time: May 20 2022 19:45:31
Neither `ifconfig` (`ifconfig -a`) nor `ip` (`ip address show`) commands are available, listing network interfaces is likely to fail
100%|███████████████████████████████████████| 338M/338M [02:10<00:00, 2.71MiB/s]
/opt/conda/envs/home-robot/lib/python3.9/site-packages/torch/functional.py:504: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /opt/conda/conda-bld/pytorch_1678402412426/work/aten/src/ATen/native/TensorShape.cpp:3483.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2023-09-11 04:01:23,363 Initializing dataset OVMMDataset-v0
2023-09-11 04:01:24,120 initializing sim OVMMSim-v0
[04:01:34:587195]:[Core] ManagedContainerBase.h(329)::checkExistsWithMessage : <Lighting Layout>::getObjectByHandle : Unknown Lighting Layout managed object handle :  . Aborting
2023-09-11 04:01:39,725 Initializing task OVMMNavToObjTask-v0
/opt/conda/envs/home-robot/lib/python3.9/site-packages/gym/spaces/box.py:84: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
Loading pretrained CLIP
  0%|          | 1/1199 [03:42<74:10:04, 222.88s/it]
```

cpaxton, 2023-09-11:
I think this is fine, the episode viewpoints changed so we would not expect the same results here. We can update the Readme if this is confusing.

(base) ***@***.***:~/home-robot/projects/habitat_ovmm$ ./scripts/test_local.sh --docker-name ovmm_baseline_submission

==========

https://developer.nvidia.com/ngc/nvidia-deep-learning-container-license

ruddradev, 2023-09-14:
Hello, I tried running the local evaluation script after updating the docker, do I need to update the dataset as well?

I got the below error - 

```

==========
== CUDA ==
==========

CUDA Version 11.7.1

Container image Copyright (c) 2016-2023, NVIDIA CORPORATION & AFFILIATES. All rights reserved.

This container image and its contents are governed by the NVIDIA Deep Learning Container License.
By pulling and using the container, you accept the terms and conditions of this license:
https://developer.nvidia.com/ngc/nvidia-deep-learning-container-license

A copy of this license is made available in this container at /NGC-DL-CONTAINER-LICENSE for your convenience.

pybullet build time: May 20 2022 19:45:31
Neither `ifconfig` (`ifconfig -a`) nor `ip` (`ip address show`) commands are available, listing network interfaces is likely to fail
/opt/conda/envs/home-robot/lib/python3.9/site-packages/torch/functional.py:504: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /opt/conda/conda-bld/pytorch_1666642975993/work/aten/src/ATen/native/TensorShape.cpp:3190.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2023-09-14 07:33:04,336 Initializing dataset OVMMDataset-v0
2023-09-14 07:33:05,232 initializing sim OVMMSim-v0
[07:33:08:192983]:[Core] ManagedContainerBase.h(329)::checkExistsWithMessage : <Lighting Layout>::getObjectByHandle : Unknown Lighting Layout managed object handle :  . Aborting
2023-09-14 07:33:10,128 Initializing task OVMMNavToObjTask-v0
/opt/conda/envs/home-robot/lib/python3.9/site-packages/gym/spaces/box.py:84: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
Loading pretrained CLIP
  0%|          | 0/1199 [00:00<?, ?it/s]Traceback (most recent call last):
  File "/home-robot/projects/habitat_ovmm/agent.py", line 97, in <module>
    metrics = evaluator.evaluate(
  File "/home-robot/projects/habitat_ovmm/evaluator.py", line 459, in evaluate
    return self.local_evaluate(agent, num_episodes)
  File "/home-robot/projects/habitat_ovmm/evaluator.py", line 259, in local_evaluate
    observations, done = self._env.reset(), False
  File "/home-robot/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py", line 116, in reset
    habitat_obs = self.habitat_env.reset()
  File "/opt/conda/envs/home-robot/lib/python3.9/site-packages/gym/core.py", line 283, in reset
    return self.env.reset(**kwargs)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/utils/gym_adapter.py", line 300, in reset
    obs = self._env.reset()
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 54, in reset
    observations = super().reset()
  File "/opt/conda/envs/home-robot/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 422, in reset
    return self._env.reset()
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 262, in reset
    observations = self.task.reset(episode=self.current_episode)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/sub_tasks/nav_to_obj_task.py", line 53, in reset
    obs = super().reset(episode)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/sub_tasks/nav_to_obj_task.py", line 268, in reset
    return self._get_observations(episode)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_task.py", line 193, in _get_observations
    task_obs = self.sensor_suite.get_observations(
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/simulator.py", line 229, in get_observations
    return Observations(self.sensors, *args, **kwargs)
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/simulator.py", line 123, in __init__
    data = [
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/simulator.py", line 124, in <listcomp>
    (uuid, sensor.get_observation(*args, **kwargs))
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/sub_tasks/nav_to_obj_sensors.py", line 118, in get_observation
    obs[..., 1] = self._get_obs_channel(
  File "/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/sub_tasks/nav_to_obj_sensors.py", line 87, in _get_obs_channel
    handle = self._sim.receptacles[
KeyError: '31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c_:0000|receptacle_mesh_31ce9e9df83a6deb8d40ab8bba2df1dfd874e36c.0000'
  0%|          | 0/1199 [00:04<?, ?it/s]
```

yvsriram, 2023-09-14:
Yes, please update the dataset.

yvsriram, 2023-09-14:
@FUIGUIMURONG we updated the docker file to run local evaluations on minival episodes instead of val. This will run it on 10 episodes instead of 1199 episodes. Please update `data/datasets/ovmm` (for getting minival) and use the updated dockerfile.

ruddradev, 2023-09-15:
Hi @yvsriram: I updated the docker and the dataset. The dataset commit is now - 9ad25fbd86a3fd352c7a0fc1f99132fbb5802378.

I am still getting this error. Could it be that the old docker container is still cached and being used? Need your advise on how to fix this.

yvsriram, 2023-09-15:
@ruddradev 

Do local evaluations (without the docker) work for you - the demo command (https://github.com/facebookresearch/home-robot/tree/main/projects/habitat_ovmm#demo-setup)? 

If yes, you could be using an old version of docker. Please pull the base image with `docker pull fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023` and rebuild your docker image.

If no, can you check if your `habitat-lab` version is up-to-date?

ruddradev, 2023-09-16:
Hi @yvsriram :

Thanks for the help. I missed the build step for the docker. When I built it now it works fine. No issues from my side here.

closed 2024-05-03 by yvsriram