facebookresearch/home-robot #372: What does one timestep correspond to?

ruddradev, 2023-09-07:
Hello,

I wanted some clarity on the concept of a timestep. As the simulator is physics enabled, I can understand that the motion will be continuous. In that respect, I found a value of timestep in /home-robot/data/default.physics_config.json which is 0.008. But when I ran the baseline for habitat-ovmm callenge evaluation, I saw consecutive frames with huge displacement between them. Does that mean the robot moves that much (20-30 degrees) in that time? Or is it an aggregation of multiple steps.

Also I assume the robot physics will not allow sharp turns such as left followed by a sharp right, as the robot inertia will play a part. Is the inertia less or disabled in the baseline? Attaching two consecutive frames from the baseline evaluation.
[image]
[image]

cpaxton, 2023-09-07:
Timesteps are arbitrary amounts of time, since we send a command to the low-level controller at each point. On the real robot they can be relatively large. There won't be any problems executing such a command on the real robot.

ruddradev, 2023-09-07:
Thank you, this clarifies the query.

closed 2023-09-07 by ruddradev