facebookresearch/home-robot #373: Could not use the “HBAO”

HusterYoung, 2023-09-07:
In the file "default_structured_configs.py", there is a factor -> hbao_visual_effects: bool. I set it to True, but the following error was reported during the evaluation process. Does anyone know how to solve it?
……
Loading pretrained CLIP
**GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidOperation**

cpaxton, 2023-09-07:
Could you please confirm for us:
- which version of habitat-sim you are using
- which version of habitat-lab you are using
- how you installed the above

HusterYoung, 2023-09-07:
> Could you please confirm for us:
> 
> * which version of habitat-sim you are using
> * which version of habitat-lab you are using
> * how you installed the above

I use habitat-lab = 0.2.3
habitat-sim = 0.2.4, and installed by pip follow the tutorial.

cpaxton, 2023-09-11:
Our custom version of habitat should support HBAO + improved rendering, but we've noticed some issues like what you're seeing.

We're working on an update - @jturner65 @aclegg3 might be able to point to a new version of habitat?