facebookresearch/home-robot #374: Can't reproduce the results of OVMM

kellyiss, 2023-09-07:
I evaluate the rl_agent with GT semantics using this command:
python projects/habitat_ovmm/eval_baselines_agent.py --baseline_config_path projects/habitat_ovmm/configs/agent/rl_agent.yaml

The prertained skill models are downloaded from https://dl.fbaipublicfiles.com/habitat/data/baselines/v1/ovmm_baseline_home_robot_challenge_2023.zip

The result is:
{'episode_count': 1200.0, 'does_want_terminate': 0.18916666666666668, 'num_steps': 1095.5791666666667, 'find_object_phase_success': 0.3258333333333333, 'pick_object_phase_success': 0.28583333333333333, 'find_recep_phase_success': 0.15166666666666667, 'overall_success': 0.0225, 'partial_success': 0.19645833333333335}

Which is lower the reported results in the paper: HomeRobot: Open-Vocabulary Mobile Manipulation.

Can you help figure out the reason?

yvsriram, 2023-09-07:
Yes, the find_object_phase_success in your case is lower than the we reported or expect with RL skill. Can you please send the `home-robot` commit you used to get these numbers?

kellyiss, 2023-09-07:
commit b643698a372c9b64af8995d7295d28ba8b7dfe99

cpaxton, 2023-09-07:
Thanks for bringing this up, we'll look into it asap

yvsriram, 2023-09-07:
Yes, this is due to a regression that got introduced in our codebase - the RL skill during eval was using a step size different from the one that was used during training. This was fixed in this commit cb6c0255c98aea24bf1d894716cd8a368b4e61d9. You should see better number when using this or the latest `main`.

kellyiss, 2023-09-10:
Since I run the evaluation process for multiple times, I found that the evaluation time can vary from 1 day to 5 days. What might cause this unstable evaluation time?

yvsriram, 2023-09-10:
Did you keep everything the same when running evaluations?

Our heuristic baseline takes longer than our RL baseline and setting PRINT_IMAGES:1 (like in `hssd_demo.yaml` config) also increases the evaluation time.

kellyiss, 2023-09-11:
I use the newest version of  home-robot and don't change the config. It took about 10 hours to process 100 episodes which is too slow. I use NVIDIA GeForce RTX 3090.

cpaxton, 2023-09-11:
Can you try making the change noted above (PRINT_IMAGES:0)

kellyiss, 2023-09-11:
https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/configs/env/hssd_eval.yaml#L6 
I think I'm using this config in which PRINT_IMAGES=0.

kellyiss, 2023-09-11:
The used gpu memory during evaluation is 1902MiB. Is it normal?

cpaxton, 2023-09-11:
Honestly this looks fine to me

cpaxton, 2023-09-12:
Yeah @kellyiss the heuristic baseline with Detic is quite slow! @yvsriram has suggested switching to groundedSAM if this is too slow for you to build on top of

kellyiss, 2023-09-13:
I evaluate the rl agent baseline again with GT semantic, this is the result:
Metrics:
 {'episode_count': 1199.0, 'does_want_terminate': 0.3803169307756464, 'num_steps': 928.1284403669724, 'find_object_phase_success': 0.5613010842368641, 'pick_object_phase_success': 0.5162635529608006, 'find_recep_phase_success': 0.41201000834028356, 'overall_success': 0.05254378648874062, 'partial_success': 0.38552960800667224}
 All seems normal except for the 'overall_success'. Is there anything wrong with this metric?

yvsriram, 2023-09-13:
Hi, 

It is possible that the numbers may be a bit different compared to what we have in the paper because our setup has changed since then. For example, the 6-point increase in FindRec success is due to an issue we fixed related to goal duplication. For the 6-point drop in overall success, I am not aware of any change that could cause this but I will check this once more. We are rerunning all evaluations with the new setup and will update the numbers in the paper this week.

asfandasfo, 2024-01-10:
@kellyiss have you solved the problem with overall success? I'm facing the same problem.

yvsriram, 2024-01-11:
Btw, we had fixed this issue in this commit 77746a741639f58975433856cfef60043802ee4c and [this comment](https://github.com/facebookresearch/home-robot/issues/374#issuecomment-1717737207) was addressed.

yvsriram, 2024-05-03:
Closing this issue hoping it has been resolved. Do let us know if you are are still facing any issues.

closed 2024-05-03 by yvsriram