facebookresearch/home-robot #390: Error while submitting docker for remote evaluation - Permission denied

ruddradev, 2023-09-21:
## 🐛 Bug

While running the remote evaluation on the baseline, using the below command - 

```
evalai push ovmm_baseline:latest --phase neurips-ovmm-minival-2023-2100 --private
```
I am getting the below error - 

```
Traceback (most recent call last):
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/urllib3/connectionpool.py", line 714, in urlopen
    httplib_response = self._make_request(
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/urllib3/connectionpool.py", line 415, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 980, in send
    self.connect()
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/docker/transport/unixconn.py", line 42, in connect
    sock.connect(self.unix_socket)
PermissionError: [Errno 13] Permission denied

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/urllib3/connectionpool.py", line 798, in urlopen
    retries = retries.increment(
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/urllib3/packages/six.py", line 769, in reraise
    raise value.with_traceback(tb)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/urllib3/connectionpool.py", line 714, in urlopen
    httplib_response = self._make_request(
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/urllib3/connectionpool.py", line 415, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/http/client.py", line 980, in send
    self.connect()
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/docker/transport/unixconn.py", line 42, in connect
    sock.connect(self.unix_socket)
urllib3.exceptions.ProtocolError: ('Connection aborted.', PermissionError(13, 'Permission denied'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/anaconda3/envs/home-robot/bin/evalai", line 8, in <module>
    sys.exit(main())
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/evalai/submissions.py", line 110, in push
    docker_image = docker_client.images.get(image)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/docker/models/images.py", line 312, in get
    return self.prepare_model(self.client.api.inspect_image(name))
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/docker/utils/decorators.py", line 19, in wrapped
    return f(self, resource_id, *args, **kwargs)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/docker/api/image.py", line 245, in inspect_image
    self._get(self._url("/images/{0}/json", image)), True
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/docker/api/client.py", line 215, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/home/anaconda3/envs/home-robot/lib/python3.9/site-packages/requests/adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', PermissionError(13, 'Permission denied'))
```

cpaxton, 2023-09-25:
Does docker run without sudo? Make sure you can run the docker hello-world example:

```
$ docker run hello-world

Hello from Docker!
This message shows that your installation appears to be working correctly.

To generate this message, Docker took the following steps:
 1. The Docker client contacted the Docker daemon.
 2. The Docker daemon pulled the "hello-world" image from the Docker Hub.
    (amd64)
 3. The Docker daemon created a new container from that image which runs the
    executable that produces the output you are currently reading.
 4. The Docker daemon streamed that output to the Docker client, which sent it
    to your terminal.

To try something more ambitious, you can run an Ubuntu container with:
 $ docker run -it ubuntu bash

Share images, automate workflows, and more with a free Docker ID:
 https://hub.docker.com/

For more examples and ideas, visit:
 https://docs.docker.com/get-started/
 ```

ruddradev, 2023-09-26:
Hi Chris,

Thanks for the help. I am not getting the error. However currently I am getting a new error when I submit the baseline. Output below.

```
(evalai) ruddra@machine:~$ evalai push ovmm_baseline_submission:latest --phase neurips-ovmm-minival-2023-2100

Error: None

(evalai) ruddra@machine:~$ evalai push ovmm_baseline_submission:latest --phase neurips-ovmm-test-standard-2023-2100

Error: None
```

cpaxton, 2023-09-26:
We are looking into this, I'm guessing its a problem with the Eval AI setup

gchhablani, 2023-09-26:
Hi @ruddradev
Can you please share these details:
- Have you set your auth token on EvalAI CLI correctly?
- Are other evalai commands working for you?
- Have you tried building evalai-cli dev version from master branch?

ruddradev, 2023-09-28:
Hi Gunjan, I tried your suggestions and getting the same error, here are my observations - 

1 - I set it and got the message "Token set successfully" so I guess it is done.
2 - I am able to login to evalai via CLI so it is working.
3 - Yes I now have built evalai-CLI from Github and using that inside a conda environment. 

If there is anything else I can help with let me know.

G-KUMAR, 2023-09-29:
> Hi Chris,
>
> Thanks for the help. I am not getting the error. However currently I am getting a new error when I submit the baseline. Output below.
>
> ```
> (evalai) ruddra@machine:~$ evalai push [...]

Getting this same error with "evalai push" . Tried everything suggested above. Any other reason this may be happening?

gchhablani, 2023-09-29:
@ruddradev @G-KUMAR I am looking into this, will get back in some time.

ruddradev, 2023-09-30:
Hi @G-KUMAR @gchhablani @cpaxton :

This issue is fixed, Apparently my token had expired. I was able to refresh the token and push the image. Possibly Evalai can put a check and throw an error when someone tries to set an expired token. Rest is fine.

Warm regards,
Ruddra

closed 2023-09-30 by ruddradev

gchhablani, 2023-09-30:
Thanks @ruddradev! Glad to know you were able to fix it. We will look into this feedback.