facebookresearch/home-robot #396: TypeError: process_instances_for_env() got an unexpected keyword argument 'semantic_seg'

gitatwork19, 2023-09-26:
## 🐛 Bug

Branch: main

The arguments passed in process_instances_for_env function call
[here](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot/home_robot/mapping/semantic/instance_tracking_modules.py#L1004-L1011):
```
self.process_instances_for_env(
      env_id,
      instance_segs[env_id],
      point_cloud[env_id],
      pose[env_id],
      image[env_id],
      semantic_seg=semantic_seg,
)
```
does not match the function implementation [here](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot/home_robot/mapping/semantic/instance_tracking_modules.py#L742-L755)
```
def process_instances_for_env(
    self,
    env_id: int,
    instance_seg: Tensor,
    point_cloud: Tensor,
    image: Tensor,
    cam_to_world: Optional[Tensor] = None,
    instance_classes: Optional[Tensor] = None,
    instance_scores: Optional[Tensor] = None,
    semantic_seg: Optional[torch.Tensor] = None,
    mask_out_object: bool = True,
    background_instance_label: int = 0,
    valid_points: Optional[Tensor] = None,
):
```
causing an `TypeError: process_instances_for_env() got an unexpected keyword argument 'semantic_seg'` error

gitatwork19, 2023-09-26:
my bad, I realised that the rest of the arguments are actually optional (from cam_to_world to valid_points), but for some reason I'm getting the above error

yvsriram, 2023-09-26:
Can you please share the full stacktrace? It seems like I am able to run evaluations on main without errors. Are you trying to use the InstanceMemory code (with `record_instance_ids=True`)? Or are you facing this error when running with default configs?

yvsriram, 2023-09-26:
It would be helpful if you could also share the command you are running.

gitatwork19, 2023-09-27:
It was using the deafult branch on the Docker container. I explicitly pull to main branch and this error's gone. Thanks @yvsriram

closed 2023-09-27 by gitatwork19