facebookresearch/home-robot #398: Missing positional error instance_memory.process_instances call function in categorical_semantic_map_module.py

gitatwork19, 2023-09-27:
## 🐛 Bug

Branch: main

There's a missing positional agument in instance_memory.process_instances function call  [here](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot/home_robot/mapping/semantic/categorical_2d_semantic_map_module.py#L515C17-L520) :
```
self.instance_memory.process_instances(
      instance_channels,
      point_cloud_t,
      image=obs[:, :3, :, :],
      semantic_channels=semantic_channels,
)
```

The function requires five arguments but only four were given as written [here](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot/home_robot/mapping/semantic/instance_tracking_modules.py#L969-L976)

```
def process_instances(
      self,
      instance_channels: Tensor,
      point_cloud: Tensor,
      pose: torch.Tensor, # <--- This is missing
      image: Tensor,
      semantic_channels: Optional[Tensor] = None,
):
```

My record_instance_ids is set to True.

Here's the whole error: https://pastebin.com/eHvMyKNd

yvsriram, 2023-09-27:
Thanks for the issue and thanks for expressing interest in using the instance memory code!

Yes, I am actually aware that InstanceMemory stuff is broken for home-robot sim on main branch due to the recent updates (probably tested with voxel code). In fact, I fixed this on a separate branch and I can easily migrate those changes to main. This was not on a critical path for us and was disabled in sim by default. But, I will start a PR for supporting this shortly.

grassjelly, 2023-09-27:
thanks @yvsriram looking forward to the PR.

yvsriram, 2023-09-29:
This PR is merged now.

closed 2023-09-29 by yvsriram

gitatwork19, 2023-10-03:
thanks @yvsriram