facebookresearch/home-robot #403: Request for Additional Resources

peryion, 2023-09-30:
## 🐛 Bug

During the evaluation of our Docker image on EvalAI, we encountered a timeout issue. To address this, we kindly request a substantial extension of the testing time to about 100 hours.

peryion, 2023-09-30:
[https://drive.google.com/file/d/1EbCx3vtce8NCLkxsLW7doSmLo75H7dKm/view?usp=drive_link](url)

peryion, 2023-09-30:
[image]

yvsriram, 2023-09-30:
Thanks, for giving the estimate of the time you need. We will check if this is possible and get back soon.

yvsriram, 2023-09-30:
We have extended it. Please try again.

peryion, 2023-10-12:
Thanks for your response. I tried again, but it still times out after running for 72 hours.
Test challenge phase and Test standard phase.

yvsriram, 2023-10-12:
@peryion can you please share your team name on EvalAI?

@gchhablani any idea why it times out after 72 hours?

peryion, 2023-10-12:
I submit my solution on Oct 8, 2023 3:41:44 PM for test challenge phase and on Oct 1, 2023 6:59:30 AM for test standard phase. But they all time out.

my team name is GVL

peryion, 2023-10-13:
When I submitted the heuristic baseline for testing, it took 60 hours, and I estimated that my solution would take around 100 hours to complete the testing. However, it timed out at 72 hours.

The testing time for my solution is approximately 1.5 times that of the heuristic baseline.

gchhablani, 2023-10-13:
@peryion I had verified that the evaluation time was 100 hours for other submissions. Can you please share if this was a fresh submission or a rerun? Can you please also share the submission ID for which you are facing this issue?

Thanks!

peryion, 2023-10-13:
It was a rerun submission.  I haven't updated my code to the latest version. Its commit is 645a338. 
Do I need to update my code, or make some changes?

Submission ID: 309068 for test standard phase, 309776 for test challenge phase.

gchhablani, 2023-10-16:
@peryion This issue is fixed.

peryion, 2023-10-17:
Thanks! I will try it again.

gchhablani, 2023-10-27:
@peryion That's strange. Does it still timeout with 72 hours?

peryion, 2023-10-27:
Yes, it timed out at 72 hours.

cpaxton, 2023-10-27:
Maybe we can re run it? Can you share submission id and all other relevant information?

cpaxton, 2023-11-22:
Closing this issue now that competition is done and the issue was resolved

closed 2023-11-22 by cpaxton