facebookresearch/home-robot #404: Unknown Lighting Layout when initializing sim

MittelmanDaniel, 2023-09-30:
## 🐛 Bug

When running test_local.sh for habitat ovmm 
It prints about an unknown lighting layout and prints the message during most of the evaluation iterations.
[image]
[image]

I am running this on a local ubuntu 22.04 install

## Steps to Reproduce

Steps to reproduce the behavior:
Follow the instructions here: https://github.com/facebookresearch/home-robot/blob/main/docs/challenge.md 
Until step 6. I used the script linked but it would stop functioning partway through and not download the entire dataset, so then when I went to manually download there'd be a conflict. I rm -rf the entire repo on my machine. Cloned it back and then manually went through each of the commands to download the dataset. Then I ran the evaluation script and achieved the output.

## Expected behavior

I expected the bar to simply grow and the number to go from 1-10 not for there to be a warning.
Additionally if there was a warning I'd expect it for all 10 iterations not for iterations 2 and 5 to not have it.

## Additional context

I apologize if this bug report isn't perfect I'm an undergraduate trying to learn.

yvsriram, 2023-09-30:
Hey, 

Thanks for participating in the challenge! These warning messages and the additional logs/steps are expected. 

@jturner65 is there a way to disable these warnings related to lighting layout?

yvsriram, 2023-09-30:
> I used the script linked but it would stop functioning partway through and not download the entire dataset

Do you know which part of the script was not working for you? Was it the dataset download part?

MittelmanDaniel, 2023-09-30:
Yes it was the dataset download part it would say something about not being able to checkout.

closed 2024-05-03 by yvsriram