facebookresearch/home-robot #417: Online and offline minival test have different results

yansenhan, 2023-10-06:
In my offline minival test, the overall success is 0.125. However, after pushing the container online, the minival test becomes 0.05.

Is there any difference between the online and offline minival test cases in OVMM competition？

cpaxton, 2023-10-06:
There should not be any difference - @yvsriram

yansenhan, 2023-10-10:
> Do you have the skill-wise successes with you? If so, can you please share?
Offline skillwise success:
overall_success: 0.0
partial_success: 0.125

Online  successs:
overall_success: 0.0
partial_success: 0.05

yansenhan, 2023-10-10:
> Also can you make sure that you're using the right environment version locally? We recently updated the image with a new version, you should specifically use the tag home-robot-ovmm-challenge-2023-v0.1.2:
> 
> ```
> git clone https://github.com/facebookresearch/home-robot --branch home-robot-ovmm-challenge-2023-v0.1.2
> ```
> 
> As indicated in the README.

The latest commit of my home-robot repository is:
commit 6eebf7af730edb176590ce1b275763109f43e788 (HEAD -&gt; main, origin/main, origin/HEAD)
Author: Sasha Sax &lt;sax@berkeley.edu&gt;
Date:   Mon Sep 18 17:09:21 2023 -0700

yansenhan, 2023-10-10:
> Also can you make sure that you're using the right environment version locally? We recently updated the image with a new version, you should specifically use the tag [...]

The latest commit of my home-robot repository is:

I find that, in docker file, the base docker image becomes "fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023-v0.2", will this have effects on the final results?

I think the online and offline docker image is the same, which should give the same results.

yvsriram, 2023-10-10:
> I find that, in the latest docker file, the base docker image becomes "fairembodied/habitat-challenge:homerobot-ovmm-challenge-2023-v0.2", will this have effects on the final results, if I still use the original base image?

Like @cpaxton mentioned, you are not supposed to use the latest `main` if participating in the challenge.

> I think the online and offline docker image is the same, which should give the same results.

Yes, but the minival dataset had changed in the local version. Please try placing 
[episodes.json.gz](https://github.com/facebookresearch/home-robot/files/12858157/episodes.json.gz) under `data/datasets/ovmm/minival/`.

closed 2024-04-11 by yvsriram