facebookresearch/home-robot #429: Unable to replicate specific episodes using episode_ids

ZealousGinger, 2023-10-12:
We found out that the OVMM dataset was recently updated and after updating the dataset, we noticed the the episodes were shuffled around. We initially kept track of the episode ids for certain specific episode to do testing, however, now that the dataset is updated and the episode were shuffled. We are unable to get the actual episode_id of an episode from the evaluator.py as it returns a different episode id.

We modified the evaluator.py file to get all episodes goal_name and episode_id as shown here
``` 
print("EPISODE INFO: ", current_episode, " ", observations.task_observations["goal_name"])
``` 

when we run
``` 
 python projects/habitat_ovmm/eval_baselines_agent.py habitat.dataset.episode_ids="[some_episode_id]"
``` 
and get the episode goal name and id printed it does not match what we entered. Some examples are shown below with the number before the arrow being the input episode_id and after arrow showing the printed details. 
``` 
0 -> 1148 Move toy_vehicle from washer_dryer to chair
22 -> 1164 Move statue from couch to cabinet
99 -> 1160 Move credit_card from table to cabinet
173 -> 462 Move sponge from chair to table
200 -> 1008 Move toy_construction_set from washer_dryer to stool 
300 -> 979 Move plant_saucer from stool to bathtub
400 -> 84 Move action_figure from chest_of_drawers to counter
462 -> 22 Move cellphone from cabinet to chest_of_drawers
500 -> 855 Move toy_vehicle from chair to stool
600 -> 509 Move hard_drive from toilet to table
``` 
 We are unsure how do we get the correct episode ids along with their respective goal names so that we can replicate specific episodes.

yvsriram, 2023-10-12:
Hi, thanks for reaching out.

I think the episodes should not be shuffled. The episodes are now retrieved by the `episode_id` in the episodes: `data/datasets/ovmm/val/episodes.json.gz`. Earlier this was not the case, so I fixed it at some point.

But if you want to match with what it was like earlier: you can try adding back this line in `src/third_party/habitat-lab/habitat-lab/habitat/datasets/ovmm/ovmm_dataset.py` - https://github.com/facebookresearch/habitat-lab/commit/46987673026ff849beba6a4b98f3ba656eb3d61b

yvsriram, 2023-10-12:
Can you share the version of episodes you were using earlier? It may also be possible that there is no correspondence between episode sets. We had regenerated the episodes at some point in time with an update to viewpoints.

ZealousGinger, 2023-10-13:
> Hi, thanks for reaching out.
> 
> The episodes are now retrieved by the `episode_id` in the episodes: `data/datasets/ovmm/val/episodes.json.gz`. Earlier this was not the case, so I fixed it at some point.
> 
> But if you want to match with what it was like earlier: you can try adding back this line in `src/third_party/habitat-lab/habitat-lab/habitat/datasets/ovmm/ovmm_dataset.py` - [facebookresearch/habitat-lab@4698767](https://github.com/facebookresearch/habitat-lab/commit/46987673026ff849beba6a4b98f3ba656eb3d61b)

Hi @yvsriram  thanks for the reply, sorry I was not clear and forgot to mention we are using this specific commit from [facebookresearch/home-robot@744b06ff6bb0c80f1417884b35de69cbc8252b6f](https://github.com/facebookresearch/home-robot/tree/744b06ff6bb0c80f1417884b35de69cbc8252b6f) which already contains the line you have mentioned. 

As for the version of episode used earlier I tried to check for the commit in the repository for ovmm_episode on huggingface but I could not find the specific commit but I know that the earlier version of episodes we downloaded it on 7 july.

yvsriram, 2024-05-03:
Not sure if you were able to figure out the mapping. We are using a stable set of episodes - so I believe you wouldn't run into these in future.

closed 2024-05-03 by yvsriram