facebookresearch/home-robot #430: placed object on receptical but overall_success == 0

LebinDing749, 2023-10-12:
## 🐛overall_success bug

The robot have palced the object on the receptical, but, after over 50 timesteps it was still be identified as 'overall_success': 0.0
I encountered several cases like this. If the success condition is not clear? Or, I meet wrong with the success condition.
 
the snapshot_pngs are here: (episode_id = 512, ground_truth = 1)
[image]
[image]
[image]

yvsriram, 2023-10-12:
Hi, thanks for creating the issue. The failures could be because of robot-scene collisions or the object not coming to rest.

Can you check these metrics?
`END.robot_collisions.robot_scene_colls`
`END.picked_object_angular_vel`
`END.picked_object_linear_vel`

LebinDing749, 2023-10-12:
Thanks,
I read the episode_result.json

"END.robot_collisions.robot_scene_colls": 2.0,
"END.picked_object_angular_vel": 0.0,
"END.picked_object_linear_vel": 0.0,

Can you explain what these parameters mean?

closed 2024-04-11 by yvsriram