facebookresearch/home-robot #431: What are the names of the objects corresponding to the values in the home_robot.core.interfaces.Observations.task_observations['gt_instance_ids']?

FUIGUIMURONG, 2023-10-16:
I want to get the object name that home_robot.core.interfaces.Observations.task_observations['gt_instance_ids'] represent.
How can I get it?

lyx-777, 2023-10-17:
Have you solved this issue? I have the same question with you.

yvsriram, 2023-10-17:
Hi

- If you are participating in the challenge - this `gt_instance_ids` instance segmentation is privileged ground truth information which you cannot assume access to. This is not being used by any of the OVMM baselines.
- These are simulator-assigned object ids. The id to name mapping is again a priviliged environment and is not exposed to the challenge environment. Accessing this information will need changes to environment (not allowed for challenge), which I will describe next.
- To answer your question, inside home-robot environment class add the following to get the object name for a `id` in gt_instance_ids

```
sim  = self.habitat_env.env.env._env.sim # access sim inside home_robot env
rom  = sim.get_rigid_object_manager() # get rigid object manager
object_handle = rom.get_object_handle_by_id(id)[:-6] # get object handle from instance id
```

Now find `object_handle` in the mappings [here](https://huggingface.co/datasets/ai-habitat/OVMM_objects/blob/main/object_categories.csv) or [here](https://huggingface.co/datasets/ai-habitat/OVMM_objects/blob/main/hssd-receptacles.csv) to get object category.

Please let me know if you have any other questions.

closed 2023-10-20 by FUIGUIMURONG

antoniodecinque99, 2023-11-05:
I am trying to create bounding boxes of the objects seen in the view. I am using the task_observations['gt_instance_ids'], as mentioned in this issue. Also, I did what you suggested here, but I am not getting the correct categories. Any help would me so appreciated. Thanks!

antoniodecinque99, 2023-11-05:
> Hi
> 
> * If you are participating in the challenge - this `gt_instance_ids` instance segmentation is privileged ground truth information which you cannot assume access to. This is not being used by any of the OVMM baselines. I believe this is not made available in challenge environment as well.
> * These are simulator-assigned object ids. The id to name mapping is again a priviliged environment and is not exposed to the challenge environment. Accessing this information will need changes to environment (not allowed for challenge), which I will describe next.
> * To answer your question, inside home-robot environment class add the following to get the object name for a `id` in gt_instance_ids
> 
> ```
> sim  = self.habitat_env.env.env._env.sim # access sim inside home_robot env
> rom  = sim.get_rigid_object_manager() # get rigid object manager
> object_handle = rom.get_object_handle_by_id(id)[:-6] # get object handle from instance id
> ```
> 
> Now find `object_handle` in the mappings [here](https://huggingface.co/datasets/ai-habitat/OVMM_objects/blob/main/object_categories.csv) or [here](https://huggingface.co/datasets/ai-habitat/OVMM_objects/blob/main/hssd-receptacles.csv) to get object category.
> 
> Please let me know if you have any other questions.

I am trying to create bounding boxes of the objects seen in the view. I am using the task_observations['gt_instance_ids'], as mentioned in this issue. Also, I did what you suggested here, but I am not getting the correct categories. Any help would me so appreciated. Thanks!

cazhang, 2024-06-12:
does this work?