facebookresearch/home-robot #433: Can't train a good "place" skill.

Jay-Ye, 2023-10-17:
## 🐛 Bug

I used the provided code to train the "place" skill with the command below. I can successfully run training, while the "ovmm_place_success" and "ovmm_placement_stability" metrics during training stay 0 all the time. I have attached the training logs of the 10-th update and the 5000-th update below.

```
#/bin/bash

export MAGNUM_LOG=quiet
export HABITAT_SIM_LOG=quiet

set -x
python -u -m habitat_baselines.run \
        benchmark/ovmm=place \
        habitat_baselines.checkpoint_folder=data/new_checkpoints/ovmm_ori_lab/place \
        --config-name ovmm/rl_skill \
```

```
update: 10	fps: 302.235	
Num updates: 10	Num frames 25600
Average window size: 10  articulated_agent_force.accum: 696.418  articulated_agent_force.instant: 378.308  ee_to_rest_distance: 0.646  force_terminate: 0.011  num_steps: 24.232  obj_anywhere_on_goal.0: 0.045  object_at_rest: 0.121  ovmm_ee_to_place_goal_distance.0: 0.971  ovmm_object_to_place_goal_distance.0: 1.355  ovmm_place_reward: -6.169  ovmm_place_success: 0.000  ovmm_placement_stability: 0.000  reward: -13.333  robot_collisions.obj_scene_colls: 0.129  robot_collisions.robot_obj_colls: 0.171  robot_collisions.robot_scene_colls: 0.751  robot_collisions.total_collisions: 1.051  robot_collisions_terminate: 0.992
	Perf Stats: trainer.rollout_collect: 8.000 trainer.sample_action: 0.002 trainer.obs_insert: 0.000 trainer.step_env: 0.056 trainer.update_stats: 0.004 trainer.update_agent: 0.463
```
```
update: 5000	fps: 441.130	
Num updates: 5000	Num frames 12800000
Average window size: 50  articulated_agent_force.accum: 1250.525  articulated_agent_force.instant: 661.172  ee_to_rest_distance: 0.787  force_terminate: 0.038  num_steps: 23.874  obj_anywhere_on_goal.0: 0.121  object_at_rest: 0.057  ovmm_ee_to_place_goal_distance.0: 0.949  ovmm_object_to_place_goal_distance.0: 0.859  ovmm_place_reward: -5.815  ovmm_place_success: 0.000  ovmm_placement_stability: 0.000  reward: -3.097  robot_collisions.obj_scene_colls: 0.059  robot_collisions.robot_obj_colls: 0.138  robot_collisions.robot_scene_colls: 0.838  robot_collisions.total_collisions: 1.035  robot_collisions_terminate: 0.967
	Perf Stats: trainer.rollout_collect: 5.359 trainer.sample_action: 0.002 trainer.obs_insert: 0.000 trainer.step_env: 0.036 trainer.update_stats: 0.004 trainer.update_agent: 0.439
```

## Steps to Reproduce

Steps to reproduce the behavior:

Here is the specific [commit](https://github.com/facebookresearch/habitat-lab/tree/9ccca099742df667c4e4fc43e204abd8c60e6028) of habitat-lab I'm using.
After setting up the environment, I simply ran the command above without any modifications to the code.

## Expected behavior

As shown the training logs attached above, the "ovmm_place_success" and "ovmm_placement_stability" metrics during training stay 0 all the time.

## Additional context

yvsriram, 2023-10-17:
Thanks for the issue. We will take a look at this, but would be helpful if you could answer these:

1. How many environments and gpus are you using?
2. Are you able to train other skills under your setup?

Jay-Ye, 2023-10-17:
Hi, thanks for your quick reply.
1. I'm using 20 environments on a single RTX 4090 GPU. (BTW, will the number of envs and gpus make a huge difference?)
2. Yes, I tried to train a "gaze" skill before with the same [rl_skill.yml](https://github.com/facebookresearch/habitat-lab/blob/6af72390cdcae262d63ab8755209b8ed8d2d9272/habitat-baselines/habitat_baselines/config/ovmm/rl_skill.yaml), the `pick_success` metric can gradually rise from ~0 to ~0.82 during the total of 100M updates. I wonder if all the skills you trained use the same config with the only difference on the `action_distribution_type`.

yvsriram, 2023-10-19:
1. Yes, they do make a difference. The policies we trained were trained with 16 envs per gpu on 2 nodes having 8 gpus each. The place skill does take a while to pick up (10M) but perfect zero success is odd. I will confirm and let you know if the issue is with the number of environments.
2. Yes, that is right.

cpaxton, 2023-11-22:
Any updates on this? Were you able to train a place skill?

@yvsriram should we update instructions on the github repo?

yvsriram, 2023-11-22:
Oh, I didn't try training it few nodes yet. Will try and get back on this.

WhyxyxIsNotAnAvailableName, 2025-07-01:
@yvsriram@Jay-Ye I ran into the same problem as you, my "ovmm_place_success" and "ovmm_placement_stability" metrics also stabilized at 0, which is also the "place" task. I also noticed that my "robot_collisions_terminate" metric was very high (staying around 0.99) and I think the success rate was 0 because of this. Do you have a solution for this, I would really appreciate it #

yvsriram, 2025-07-01:
Hi, 

> my "robot_collisions_terminate" metric was very high (staying around 0.99) and I think the success rate was 0 because of this

We disable collision termination and collision penalty in the first stage of training with: habitat.task.measurements.ovmm_place_reward.robot_collisions_pen=0.0 habitat.task.measurements.ovmm_place_reward.robot_collisions_end_pen=0.0 habitat.task.measurements.robot_collisions_terminate.max_num_collisions=-1