facebookresearch/home-robot #436: Submitted image got timed out after 71 hours of execution

ruddradev, 2023-10-20:
## 🐛 Bug

I had submitted an entry in the test-challenge phase which got timed out due to some reason. The total execution time showed as 72 hours which is less than the specified 100 hours. 

## Steps to Reproduce

Steps to reproduce the behavior:

Evalai Submission ID - 310138

## Expected behavior

Submission should run for 100 hours.

## Additional context

cpaxton, 2023-10-23:
Thanks for the comment. Looking into this

gchhablani, 2023-10-23:
@rudradev

There was some issue with some of submissions which were taking the previous specified limit (72 hours). I will check if I can re-run this submission and get back.

cpaxton, 2023-10-24:
@gchhablani can you confirm we are re-running this?

gchhablani, 2023-10-24:
@cpaxton @ruddradev Yes, we are re-running the submission.

cpaxton, 2023-10-24:
In that case closing this issue. Good luck @ruddradev

closed 2023-10-24 by cpaxton