facebookresearch/home-robot #437: This repository is over its data quota

kellyiss, 2023-10-24:
when I install 
 # Home robot dependency for velocity controllers
home-robot @ git+ssh://git@github.com/facebookresearch/home-robot.git@habitat-challenge-2023#subdirectory=src/home_robot

I got this error. What can I do to resolve it?
 Resolved ssh://****@github.com/facebookresearch/home-robot.git to commit c5b5e25ce402cce44b7315f5303e941ea9a3d252
 Running command git submodule update --init --recursive -q
 Downloading habitat_stretch/meshes/base_link.obj (594 KB)
 Error downloading object: habitat_stretch/meshes/base_link.obj (a41840c): Smudge error: Error downloading habitat_stretch/meshes/base_link.obj (a41840c67f1c7ab0069c9d95748bd0122682e9f026b3d35dbc18423c7c5d5435): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

cpaxton, 2023-10-24:
I think this is on my end, i need to move this file out of my personal github account. Will do so asap.

cpaxton, 2023-10-24:
I will provide a zip file for now

cpaxton, 2023-10-24:
For now, please clone the other submodules individually

cpaxton, 2023-10-24:
@kellyiss it was added here: https://github.com/facebookresearch/home-robot/tree/main/assets/hab_stretch

Since it's fixed on main, I think I will close this issue - although @yvsriram maybe we should update this in the challenge branch if we expect it to be longer lived?

cpaxton, 2023-10-24:
For now please copy these files from main

closed 2023-10-24 by cpaxton

kellyiss, 2023-10-25:
Thanks for your help.

sai-prasanna, 2023-11-03:
@cpaxton I am trying to install the challenge branch of habitat and ran into this error, should I edit the extra requires of habitat-lab package to point to main branch of home-robot?