facebookresearch/home-robot #442: Recording Camera Poses

user432, 2023-11-02:
Hello, I'm trying to record some camera pose trajectories of multiple HSSD scenes from [here](https://github.com/facebookresearch/home-robot/blob/4b3c1e1bfa9e7218d3c5e09ee257f05c57c9be8d/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py#L156). When I try to use the same trajectory to render frames in Blender, it doesn't seem to work. I would like to know if there's any transformation matrix I'm missing here. Thanks!

yvsriram, 2023-11-02:
Thanks for the issue @user432! Yes, we are fixing this to be relative to the agent's start position in the episode so you can use it without further transformation for your purpose. I will let you know once it's done.

user432, 2023-11-04:
Thanks a lot @yvsriram, it worked. Closing this issue

closed 2023-11-04 by user432