facebookresearch/home-robot #444: How to retrieve object name from the instance_id inside task_observations['gt_instance_ids']?

antoniodecinque99, 2023-11-05:
I am trying to capture some images from the robot exploration in the scene. I want to create bounding boxes and object categories for each element. By using observations.task_observations["gt_instance_ids"] I correctly get the bounding boxes. However, such ids do not carry a mapping towards the object/receptacle class. 

I basically have the same issue which was highlighted here: https://github.com/facebookresearch/home-robot/issues/431
I followed the solution, but I am not getting the correct classes. Any help would be appreciated, thanks!

antoniodecinque99, 2023-11-16:
I am finding objects which do not have a correct mapping to the task classes. Here's an example

[image]

In the ground truth I have objects with classes such as "magazine" or "potted_plant". I am doing evaluation of Detic and Grounded SAM to understand how to make such models better in the overall agent. How should I evaluate, for example, "potted_plant" if such class is not present in the complete vocabulary of task classes? Detic for example predicts "plant_container", which could be considered correct. How do I map such ground truths if they don't have an exact match in the vocabulary used for the task?

yvsriram, 2024-05-03:
You need to map the object handles you get from the simulator to their categories. You can find the handle to category mapping [here](https://huggingface.co/datasets/ai-habitat/OVMM_objects/blob/main/object_categories.csv) and [here](https://huggingface.co/datasets/ai-habitat/OVMM_objects/raw/main/hssd-receptacles.csv)

closed 2024-05-27 by yvsriram