facebookresearch/home-robot #445: Could Not find ovmm/ovmm_eval.yaml for eval_baselines_agent.py

IDoMathEveryDay, 2023-11-06:
After installing habitat-sim and home-robot, I try to run the eval step by calling:
`
python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml
`
However, in the  eval_baselines_agent.py, line 35 to line 40, it requires a file called "ovmm/ovmm_eval.yaml" which cannot be found in the project

`    parser.add_argument(
        "--habitat_config_path",
        type=str,
        default="ovmm/ovmm_eval.yaml",
        help="Path to config yaml",
    )
`
Could I know where I could find this ovmm_eval.yaml file? Thanks a lot.

yvsriram, 2023-11-15:
Hi, this file should be present in the path `src/third_party/habitat-lab/habitat-baselines/habitat_baselines/config/ovmm/ovmm_eval.yaml`. It may be possible that you are on the wrong habitat-lab commit. Can you update the habitat-lab submodule so that you use the habitat-lab version tagged on `main`?

cpaxton, 2023-11-22:
@yvsriram can you share the right version of habitat lab to use?

yvsriram, 2023-11-22:
@IDoMathEveryDay, it looks like you are using habitat-lab main, where this config is not present. Please try the commit `b727ca9f7123101aaedb737ca9ccc1b153525dd9` of habitat-lab tagged on home-robot main

closed 2024-04-11 by yvsriram