facebookresearch/home-robot #448: Multiple issues with the simple_grasp_server

benoit-robotics, 2023-11-21:
I am using the home-robot-ovmm-challenge-2023-v0.1.2 tag, and I am testing the OVMM example script with the simple grasp server.
`python src/home_robot_hw/home_robot_hw/nodes/simple_grasp_server.py`

The simple grasp server crashes when the robot starts to pick up an object : 

> [ERROR] [1700422353.593442, 1700420338.783554]: Error processing request: rgb.ndim=2
> ['Traceback (most recent call last):\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 633, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/benoit/code/home-robot/src/home_robot_hw/home_robot_hw/ros/grasp_helper.py", line 184, in process\n    grasps, scores, in_base_frame = self.handle_request_fn(\n', '  File "/home/benoit/code/home-robot/src/home_robot/home_robot/manipulation/voxel_grasps.py", line 241, in get_grasps\n    voxel_map.add(camera_pose, pc_segment, feats=pc_color_segment)\n', '  File "/home/benoit/code/home-robot/src/home_robot/home_robot/mapping/voxel/voxel.py", line 259, in add\n    assert rgb.ndim == 3, f"{rgb.ndim=}"\n', 'AssertionError: rgb.ndim=2\n']

The problem is there : https://github.com/facebookresearch/home-robot/blob/home-robot-ovmm-challenge-2023-v0.1.2/src/home_robot/home_robot/mapping/voxel/voxel.py#L259 and the calling function is here : https://github.com/facebookresearch/home-robot/blob/home-robot-ovmm-challenge-2023-v0.1.2/src/home_robot/home_robot/manipulation/voxel_grasps.py#L241

I think there are multiple problems in the _add_ function of _voxel.py_ : _rgb_ is a matrix ( ndim == 2 ) so the asset fails. Just below
```
assert (
            rgb.shape[:-1] == depth.shape
        )
```
will also fail because depth is not set by the calling function and is None. In the calling function, in voxel_grasp.py, I assume that pc_segment isn't correct built because it doesn't have the right dimensions.
Any thoughts on how I could proceed ? Thank you.

cpaxton, 2023-11-22:
Hey Benoit,

Unfortunately this is a known issue on hardware, it's fixed in this branch: https://github.com/facebookresearch/home-robot/pull/435

Best,
Chris

cpaxton, 2023-11-22:
@Jdvakil has been using this branch on the real robot, and it should work

Jdvakil, 2023-11-22:
> @Jdvakil has been using this branch on the real robot, and it should work
I am using this branch and I have had no crashes

benoit-robotics, 2023-11-23:
Thank you, I will give it a try.

benoit-robotics, 2023-11-23:
@cpaxton @Jdvakil 
Before using this branch I was using v0.1.1. On v0.1.1, with
`python projects/real_world_ovmm/eval_episode.py`
the robot was able to detect, reach a cup and attempt to pick it up.

Now with the _demo-refactor_ branch the robot is not enable to select a goal, even if cups are visible and correctly identified. The robot just keeps exploring indefinitely. In the output I always get 

> Found reachable goal: False

 and 

> Found goal: False

I think that on v0.1.1 when one of these variable was true, it meant that a cup was identified and that the robot was trying to reach it. It doesn't happen here with the new branch.

cpaxton, 2023-11-23:
We have been testing this branch ourselves. Can you make sure it's not just a configuration file change by testing projects/real world ovmm/tests/test_heuristic_policies?

This will run the heuristic (not rl) policies and has command line arguments for setting goals. It should work.

cpaxton, 2023-11-23:
This output does seem to indicate that it's not finding the objects though.

benoit-robotics, 2023-11-24:
@cpaxton 
I was able to run `python projects/real_world_ovmm/tests/test_heuristic_policies.py --test-pick` successfully. However I have noticed that with this script, the base of the robot only moves along the x-axis, it means that if for instance the object to grasp is too far, the robot isn't able to move close enough to the object to reach it.
What is your approach to make the robot come close enough to the object to grasp ?

cpaxton, 2024-01-11:
This should now be resolved in main

closed 2024-01-11 by cpaxton