facebookresearch/home-robot #451: depth estimation models

xieodu, 2023-12-18:
Hello!Thanks for your work!
After reading your paper, I am interested in the implementation of the depth estimation section discussed in " Since depth estimation models typically predict relative distances instead of absolute distances, we ground the predicted depth using the known true depth values from Dt. Specifically, we solve for the scale factor that minimizes the mean squared error between estimated depth and sensed depth across all pixels where there is a depth reading."  Could you please tell me which part of the project code corresponds to this section?

cpaxton, 2023-12-19:
I believe there is depth fixing code here: https://github.com/facebookresearch/home-robot/blob/demo-refactor/src/home_robot_spot/home_robot_spot/spot_client.py#L485

although we tried several things - this part is specifically from GOAT which is still WIP

xieodu, 2023-12-20:
@cpaxton Thanks for your help! Looking forward to your new progress!

yvsriram, 2024-05-03:
Closing. Let us know if you have more questions.