facebookresearch/home-robot #453: Inquiry about the Release Schedule for Instance Memory and Mask R-CNN Integration

xuboshen, 2023-12-20:
Hi, thank you for your wonderful work!

I have observed that the segmentation algorithm detic does not perform well in the HabitatObjNav task.

I would like to inquire about the anticipated release schedule for the integration of <Instance Memory> and the incorporation of code related to <Mask R-CNN> in your project.

Thank you for your time and efforts in maintaining this project.

cpaxton, 2024-01-17:
A version of instance memory has already been released actually! The version from the GOAT paper is work in progress, the lead authors of that paper can comment on progress. @mukulkhanna can comment?

I don't think we're planning to add mask-rcnn. I think the set of classes in habitat obj nav is very restricted.

yvsriram, 2024-04-11:
Closing this issue. Please let us know if you have any more questions on the instance memory.

closed 2024-04-11 by yvsriram