facebookresearch/home-robot #459: Can't reproduce OVMM result

asfandasfo, 2024-01-06:
Hi, I following these instructions (https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md) to evaluate the rl_agent with GROUND_TRUTH_SEMANTICS: 0.
here are the results:
Metrics:
 {'episode_count': 1199.0, 'does_want_terminate': 0.05337781484570475, 'num_steps': 1217.8340283569642, 'find_object_phase_success': 0.195162635529608, 'pick_object_phase_success': 0.09841534612176814, 'find_recep_phase_success': 0.0567139282735613, 'overall_success': 0.0016680567139282735, 'partial_success': 0.08798999165971642}

find_object_phase_success look similar as report in paper but other results are very low- Am I missing something? Can you help? Thanks

yvsriram, 2024-01-06:
Hi, yes, the success numbers appear to be lower. I will check if there is an issue. It would be helpful if you could point me to the commit you are on.

asfandasfo, 2024-01-06:
this one [4b3c1e1](https://github.com/facebookresearch/home-robot/commit/4b3c1e1bfa9e7218d3c5e09ee257f05c57c9be8d)

yvsriram, 2024-01-11:
Hi, I reran evaluations with GT and Detic on 77746a741639f58975433856cfef60043802ee4c (which is close to the version you are using, any changes between the two shouldn't affect the evaluations) and was able to reproduce the [results](https://openreview.net/pdf?id=b-cto-fetlz). But I will also confirm that the latest main has no issues.

Can you please confirm if you are using the right checkpoints: did you download them using the install_deps.sh script from [here](https://dl.fbaipublicfiles.com/habitat/data/baselines/v1/ovmm_baseline_home_robot_challenge_2023_v0.2.zip
)? Also, what `habitat-lab` commit (under `src/third_party/habitat-lab` were you using?

asfandasfo, 2024-01-11:
I did evaluate it with GT perception and got these results:
[image]
I have the same lower overall_success as in #374 
I am using [`10a6e87`](https://github.com/facebookresearch/habitat-lab/commit/d965bd0264f857845d3d0167806ff17b994c6441) for `habitat-lab`.
I will try the latest `main` to see if I could reproduce the results.

yvsriram, 2024-01-11:
Just to confirm, you used the same set of commits in both evals ([GT semantics](https://github.com/facebookresearch/home-robot/issues/459#issuecomment-1887821208) and [Detic](https://github.com/facebookresearch/home-robot/issues/459#issue-2068624575)), right? 

Actually, I would expect to see a higher `find_object_phase_success` too with GT semantics when using those commits (which are post 0.2.5 upgrade). These do resemble the numbers reported in the other issue (which was before 0.2.5 upgrade). There is a possibility that your environment has older versions pre-installed even though your code is on some recent commit. Eg: Can you once double-check that the eval environment uses the right habitat version? Activate the conda environment and run `python3 -c "import habitat; print(habitat.__version__)"`

asfandasfo, 2024-01-12:
Yes, I'm using the same commits for both evals. And you are right, I have installed the older version of the habitat. Thanks for pointing out. Probably that is the reason for the lower success numbers.

I was also trying to install the whole setup to a new environment and following the same steps as before but I'm getting the this error. Can you help?
`Requirement already satisfied: numpy in /home/ayaar/anaconda3/envs/ay_home-robot/lib/python3.9/site-packages (from home-robot-sim==0.1.0) (1.23.5)
Requirement already satisfied: home-robot in ./src/home_robot (from home-robot-sim==0.1.0) (0.1.0)
INFO: pip is looking at multiple versions of home-robot-sim to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement habitat-sim (from home-robot-sim) (from versions: none)
ERROR: No matching distribution found for habitat-sim`

cazhang, 2024-01-18:
Got similar numbers, evaluating the release challenge code:  GT + RL
==================================================
Averaged metrics
==================================================
episode_count: 1199.0
does_want_terminate: 0.3769808173477898
num_steps: 930.7372810675563
find_object_phase_success: 0.5629691409507923
pick_object_phase_success: 0.5262718932443703
find_recep_phase_success: 0.43119266055045874
overall_success: 0.0633861551292744
partial_success: 0.3959549624687239
==================================================

yvsriram, 2024-04-11:
Hi @cazhang, please refer to this [comment](https://github.com/facebookresearch/home-robot/issues/459#issuecomment-1887954392). I'd suggest making sure that you are on the updated commit.

yvsriram, 2024-05-03:
I am assuming this issue is resolved. Please reopen if you are still facing issues.

closed 2024-05-03 by yvsriram