facebookresearch/home-robot #462: Installing pip dependencies: Failed

vaibhavoutat, 2024-01-10:
## 🐛 Bug

CondaEnvException: Pip failed
I have mamba and conda installed on my system but when i create the environment using mamba the following error occurs.

Steps to reproduce the behavior:
mamba env create -n home-robot -f src/home_robot/environment.yml

## Additional context
(Log of the error)
Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Installing pip dependencies: / Ran pip subprocess with arguments:
['/home/deepbot/miniforge3/envs/home-robot/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt', '--exists-action=b']
Pip subprocess output:
Collecting git+https://github.com/openai/CLIP.git (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 26))
  Cloning https://github.com/openai/CLIP.git to /tmp/pip-req-build-b04bnx8p
  Resolved https://github.com/openai/CLIP.git to commit a1d071733d7111c9c014f024669f959182114e33
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting numpy<1.24 (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 1))
  Using cached numpy-1.23.5-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.1 MB)
Requirement already satisfied: scipy in /home/deepbot/miniforge3/envs/home-robot/lib/python3.9/site-packages (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 2)) (1.11.4)
Collecting sophuspy (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 3))
  Using cached sophuspy-0.0.8-cp39-cp39-linux_x86_64.whl
Collecting pybullet (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 4))
  Using cached pybullet-3.2.6-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (1.8 kB)
Collecting trimesh (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 5))
  Using cached trimesh-4.0.8-py3-none-any.whl.metadata (18 kB)
Collecting pytest (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 6))
  Using cached pytest-7.4.4-py3-none-any.whl.metadata (7.9 kB)
Collecting scikit-image (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 7))
  Using cached scikit_image-0.22.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (13 kB)
Collecting scikit-fmm (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 8))
  Using cached scikit_fmm-2023.4.2-cp39-cp39-linux_x86_64.whl
Requirement already satisfied: scikit-learn in /home/deepbot/miniforge3/envs/home-robot/lib/python3.9/site-packages (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 9)) (1.3.2)
Collecting numpy-quaternion (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 10))
  Using cached numpy_quaternion-2023.0.2-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (14 kB)
Collecting natsort (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 11))
  Using cached natsort-8.4.0-py3-none-any.whl.metadata (21 kB)
Collecting frozendict (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 12))
  Using cached frozendict-2.4.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (23 kB)
Collecting opencv-python (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 13))
  Using cached opencv_python-4.9.0.80-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (20 kB)
Collecting pin>=2.6.17 (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 14))
  Using cached pin-2.6.21-1-cp39-cp39-manylinux_2_28_x86_64.whl.metadata (20 kB)
Collecting openai-clip (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 15))
  Using cached openai_clip-1.0.1-py3-none-any.whl
Collecting timm (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 16))
  Using cached timm-0.9.12-py3-none-any.whl.metadata (60 kB)
Collecting pandas (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 17))
  Using cached pandas-2.1.4-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (18 kB)
Collecting mss (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 18))
  Using cached mss-9.0.1-py3-none-any.whl (22 kB)
Collecting matplotlib (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 19))
  Using cached matplotlib-3.8.2-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (5.8 kB)
Collecting click (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 20))
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Requirement already satisfied: yacs in /home/deepbot/miniforge3/envs/home-robot/lib/python3.9/site-packages (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 21)) (0.1.8)
Collecting h5py (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 22))
  Using cached h5py-3.10.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.5 kB)
Collecting imageio (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 23))
  Using cached imageio-2.33.1-py3-none-any.whl.metadata (4.9 kB)
Collecting pygifsicle (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 24))
  Using cached pygifsicle-1.0.7-py3-none-any.whl
Collecting pynput (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.okgy6y9s.requirements.txt (line 25))
  Using cached pynput-1.7.6-py2.py3-none-any.whl (89 kB)
[... 195 lines cut ...]
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-zoyn90re/h5py_bd1b121e0db04141bc28cfead6694682/setup.py", line 24, in <module>
          from distutils.command.build_py \
      ImportError: cannot import name 'build_py_2to3' from 'distutils.command.build_py' (/home/deepbot/miniforge3/envs/home-robot/lib/python3.9/site-packages/setuptools/_distutils/command/build_py.py)
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

failed

CondaEnvException: Pip failed

Can you kindly tell me what is wrong?

cpaxton, 2024-01-10:
This is definitely a version issue, it looks like openai clip is what's failing? if you comment that out from the environment does everything else work?

cpaxton, 2024-01-10:
It's also using the wrong version of h5py

vaibhavoutat, 2024-01-11:
> This is definitely a version issue, it looks like openai clip is what's failing? if you comment that out from the environment does everything else work?

No even after commenting out that line I still have import error.
ImportError: cannot import name 'build_py_2to3' from 'distutils.command.build_py' (/home/deepbot/miniforge3/envs/home-robot/lib/python3.9/site-packages/setuptools/_distutils/command/build_py.py)
      [end of output]

And when I activate the  created environment (home-robot)
pip install openai-clip works fine.
Reference: https://github.com/openai/whisper/discussions/1645

vaibhavoutat, 2024-01-11:
> It's also using the wrong version of h5py

I tried to fix the import error ImportError: cannot import name 'build_py_2to3' from 'distutils.command.build_py' by manually installing setuptools<=57.5.0 in the environment.
Reference : https://stackoverflow.com/questions/75085575/importerror-cannot-import-name-build-py-2to3-from-distutils-command-build-py

But after that open3d and matplotlib can't be resolved together. Here is a snip from the error log.
Collecting open3d (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 26))
  Using cached open3d-0.14.1-cp39-cp39-manylinux_2_27_x86_64.whl (395.6 MB)
Collecting imageio (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 23))
  Using cached imageio-2.33.0-py3-none-any.whl.metadata (4.9 kB)
INFO: pip is still looking at multiple versions of cmeel-boost to determine which version is compatible with other requirements. This could take a while.
  Using cached imageio-2.32.0-py3-none-any.whl.metadata (4.9 kB)
INFO: This is taking longer than usual. You might need to provide the dependency resolver with stricter constraints to reduce runtime. See https://pip.pypa.io/warnings/backtracking for guidance. If you want to abort this run, press Ctrl + C.
  Using cached imageio-2.31.6-py3-none-any.whl.metadata (4.6 kB)
  Using cached imageio-2.31.5-py3-none-any.whl.metadata (4.6 kB)
  Using cached imageio-2.31.4-py3-none-any.whl.metadata (4.6 kB)
  Using cached imageio-2.31.3-py3-none-any.whl.metadata (4.6 kB)
  Using cached imageio-2.31.2-py3-none-any.whl.metadata (4.7 kB)
  Using cached imageio-2.31.1-py3-none-any.whl.metadata (4.7 kB)
  Using cached imageio-2.31.0-py3-none-any.whl.metadata (4.7 kB)
  Using cached imageio-2.30.0-py3-none-any.whl.metadata (4.7 kB)
  Using cached imageio-2.29.0-py3-none-any.whl.metadata (4.7 kB)
  Using cached imageio-2.28.1-py3-none-any.whl (3.4 MB)
  Using cached imageio-2.28.0-py3-none-any.whl (3.4 MB)
  Using cached imageio-2.27.0-py3-none-any.whl (3.4 MB)
Collecting h5py (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 22))
  Using cached h5py-3.9.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.5 kB)
  Using cached h5py-3.8.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.7 MB)
  Using cached h5py-3.7.0-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (4.5 MB)
  Using cached h5py-3.6.0-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (4.5 MB)
  Using cached h5py-3.5.0-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (4.5 MB)
  Using cached h5py-3.4.0-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (4.5 MB)
  Using cached h5py-3.3.0-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (4.5 MB)
  Using cached h5py-3.2.1-cp39-cp39-manylinux1_x86_64.whl (4.5 MB)
  Using cached h5py-3.2.0-cp39-cp39-manylinux1_x86_64.whl (4.5 MB)
  Using cached h5py-3.1.0-cp39-cp39-manylinux1_x86_64.whl (4.4 MB)
  Using cached h5py-3.0.0-cp39-cp39-manylinux1_x86_64.whl (4.4 MB)
Collecting cached-property (from h5py->-r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 22))
  Using cached cached_property-1.5.2-py2.py3-none-any.whl (7.6 kB)
Collecting h5py (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 22))
  Using cached h5py-2.10.0.tar.gz (301 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
  Using cached h5py-2.9.0.tar.gz (287 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
  Using cached h5py-2.8.0.tar.gz (274 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
  Using cached h5py-2.7.1.tar.gz (264 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
  Using cached h5py-2.7.0.tar.gz (256 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
  Using cached h5py-2.6.0.tar.gz (245 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
  Using cached h5py-2.5.0.tar.gz (684 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting Cython>=0.17 (from h5py->-r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 22))
  Using cached Cython-3.0.8-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.2 kB)
Collecting h5py (from -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 22))
  Using cached h5py-2.4.0.tar.gz (172 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
  Using cached h5py-2.3.1.tar.gz (1.1 MB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
[... 86 lines cut ...]
    open3d 0.17.0 depends on matplotlib>=3
    The user requested matplotlib
    open3d 0.18.0 depends on matplotlib>=3
    The user requested matplotlib
    open3d 0.16.0 depends on matplotlib>=3
    The user requested matplotlib
    open3d 0.18.0 depends on matplotlib>=3
    The user requested matplotlib
[... cut ...]
2. remove package versions to allow pip attempt to solve the dependency conflict

Pip subprocess error:
ERROR: Cannot install -r /home/deepbot/habitat/homorobottest/home-robot/src/home_robot/condaenv.cuh9ey21.requirements.txt (line 26), matplotlib and open3d==0.18.0 because these package versions have conflicting dependencies.
ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

failed

CondaEnvException: Pip failed

vaibhavoutat, 2024-01-11:
@cpaxton Another thing I noticed in the environmen.yml file that numpy <1.24 is specified along with pin>=2.6.17.
While manually installing pin and numpy , I saw that pin>=2.6.17 needs numpy>1.26.

cpaxton, 2024-01-11:
Looking into this. I think some of our unfixed packages keep causing errors every time i fix this

cpaxton, 2024-01-11:
@vaibhavoutat maybe this is because of Mamba, but when i do this i see:
```
(base) cpaxton@cpaxton-ThinkStation-P620:~/src/home-robot$ conda activate tmp03
(tmp03) cpaxton@cpaxton-ThinkStation-P620:~/src/home-robot$ pip show numpy
Name: numpy
Version: 1.23.5
Summary: NumPy is the fundamental package for array computing with Python.
Home-page: https://www.numpy.org
Author: Travis E. Oliphant et al.
Author-email: 
License: BSD
Location: /home/cpaxton/miniconda3/envs/home_robot/envs/tmp03/lib/python3.9/site-packages
Requires: 
Required-by: cmeel-boost, contourpy, fvcore, h5py, imageio, matplotlib, numpy-quaternion, open3d, opencv-python, pandas, pyquaternion, scikit-fmm, scikit-image, scikit-learn, SciPy, sophuspy, tifffile, torch_geometric, torchvision, trimesh
(tmp03) cpaxton@cpaxton-ThinkStation-P620:~/src/home-robot$ mamba env create -n tmp03 -f src/home_robot/environment.yml ^C
(tmp03) 130 cpaxton@cpaxton-ThinkStation-P620:~/src/home-robot$ pip show pin
Name: pin
Version: 2.6.20
Summary: A fast and flexible implementation of Rigid Body Dynamics algorithms and their analytical derivatives
Home-page: https://github.com/stack-of-tasks/pinocchio
Author: 
Author-email: 
License: 
Location: /home/cpaxton/miniconda3/envs/home_robot/envs/tmp03/lib/python3.9/site-packages
Requires: cmeel, cmeel-boost, cmeel-console-bridge, cmeel-tinyxml, cmeel-urdfdom, hpp-fcl
Required-by: 
```
And it works fine

cpaxton, 2024-01-11:
Environment created with 
```
mamba env create -n tmp03 -f src/home_robot/environment.yml
```

cpaxton, 2024-01-11:
can you share what  you ran?

cpaxton, 2024-01-11:
maybe my enviromment is messy somehow

vaibhavoutat, 2024-01-12:
> Environment created with
> 
> ```
> mamba env create -n tmp03 -f src/home_robot/environment.yml
> ```

I also created the environmnet with this command.

mamba env create -n home-robot -f src/home_robot/environment.yml

vaibhavoutat, 2024-01-12:
> @vaibhavoutat maybe this is because of Mamba, but when i do this i see:
>
> ```
> (base) cpaxton@cpaxton-ThinkStation-P620:~/src/home-robot$ conda activate tmp03
> (tmp03) [...]

Well I did the installation manually for all the pip dependencies and followed all the instructions ahead , somehow my numpy also shows as Version: 1.23.5 with correct pin too. Maybe it got fixed will following the further details , but I can't still make the  
mamba env create -n tmp03 -f src/home_robot/environment.yml
command to work without errors.  Same that I posted originally.

cpaxton, 2024-01-18:
So I believe this is an issue with setuptools/distutils: https://stackoverflow.com/questions/75085575/importerror-cannot-import-name-build-py-2to3-from-distutils-command-build-py

Can you share your versions of these? Maybe we need to pin a specific version to make this work better @vaibhavoutat

vaibhavoutat, 2024-02-22:
> Hey sorry for following up late , but I found out that when I install the dependencies using ovmm-challenge as suggested in the readme , it is then when I get this error, and when i use the master branch then I don't run up into the errors.
> 
> This gives error: Important note for new OVMM challenge participants: If you are participating in the challenge and starting now, please use the home-robot release home-robot-ovmm-challenge-2023-v0.1.2. You can start out by cloning home-robot via: git clone https://github.com/facebookresearch/home-robot --branch home-robot-ovmm-challenge-2023-v0.1.2
> 
> On the other hand the main branch works fine: git clone https://github.com/facebookresearch/home-robot.git
> 
> So maybe it's the difference between these files.

@cpaxton Can you please have a look.

vaibhavoutat, 2024-02-22:
> @cpaxton Can you please have a look.

I see that pinocchio is missing in the dependencies and there are stricter constraints

cpaxton, 2024-02-28:
Sorry can you switch to main? I need to change the instructions to not use the OVMM challenge branch any more; I'm not sure that's being maintained now that the challenge is over.

vaibhavoutat, 2024-02-29:
> Sorry can you switch to main? I need to change the instructions to not use the OVMM challenge branch any more; I'm not sure that's being maintained now that the challenge is over.

Okay will do that. Thanks

closed 2024-03-26 by yvsriram